PX4/PX4-Autopilot #10826: EKF GPS fusion timeout - reset to GPS

roseyangyu, 2018-11-12:
Version：
FW：1.8.0
FC：v2
gps：RTK
I have set EKF2_HGT_MODE to gps.But when I test RTK,it warnning EKF GPS fusion timeout - reset to GPS  and EKF GPS fusion timeout - reset to baro.How do I solve this warnning?

xdwgood, 2018-11-13:
Suggest: you can see PX4/ECL and find `EKF GPS fusion timeout - reset to baro`
Please aslo provide log link. 
https://logs.px4.io/plot_app?log=279de7aa-6a7b-408c-9ab9-a9a6b9e5fae0

I am also interested this issue,so commented on it.
I noticed this: 
[image]

@mhkabir  @priseborough  Could you have a look ? thanks :smiley:

roseyangyu, 2018-11-26:
I have solve this problem when I set rate to 5HZ

closed 2018-11-26 by roseyangyu