PX4/PX4-Autopilot #10834: LED driver TCA62724FMG  stop production

jamming, 2018-11-13:
TOSHIBA LED driver TCA62724FMG stop production right now. I2C controlled RGB LED driver NCP5623BMUTBG of Onsemi is good substitute for TCA62724FMG. It's a good idea to change the drivers to support it.
https://github.com/jamming/image/blob/master/NCP5623B-LED-driver.png

tubeme, 2018-11-13:
Actually we created a LED with another driver chip. NXP one. The first PCB is ready to be tested. We can share it. We new the TOSH stopped the production, because we wanted to make our LED with this chip so there were only stocks left here and there one year ago I suppose not anymore.

dagar, 2018-11-29:
A driver for the ncp5623c was recently contributed. https://github.com/PX4/Firmware/pull/10889

closed 2018-11-29 by dagar