PX4/PX4-Autopilot #10835: Multiple VTOL vehicles

IssoufGit, 2018-11-13:
Using this [example](https://dev.px4.io/en/simulation/multi-vehicle-simulation.html) I can run multiple quadcopter simulation in gazebo.

However I am looking for a way to add **multiple instances of the "standard_vtol"** in gazebo and be able to connect them to QgroundControl. 

As an attempt, I have tried to reproduce the xacro files for the standard_vtol by changing the [xacro files for iris quadcopter](https://github.com/PX4/sitl_gazebo/tree/3d80f63562c24c1537e1f8423ce649c99ebc15ea/models/rotors_description/urdf), However they are still using the dynamic of Quadcopter. 

How can I add the dynamic of VTOL to my xacro files? 
Is there any other way to add multiple VTOL into the gazebo simulation and connect them to QGroundControl?

Any help is more than welcome !

closed 2019-02-25 by stale[bot]

Mavrikant, 2019-12-27:
:( I have the same problem. There should be an easy solution. This must be a common problem for all.

Jaeyoung-Lim, 2019-12-27:
@Mavrikant This is WIP

Jaeyoung-Lim, 2019-12-29:
@Mavrikant @IssoufGit Here is a example for enabling fixed wing: https://github.com/PX4/Firmware/pull/13806

Jaeyoung-Lim, 2020-01-19:
@Mavrikant @IssoufGit Fixed in https://github.com/PX4/Firmware/pull/13985

Jaeyoung-Lim, 2020-01-19:
@Mavrikant @IssoufGit Support for vtol has been added in #13985