PX4/PX4-Autopilot #10836: Does the loiter_unlim waypoint loiter radius set not working?

goodman110110110, 2018-11-13:
When I use the LOITER_UNLIM waypoint, I set the loiter radius as 100m.
[image]
The plane will loiter around the waypoint in the mission mode, but the loiter radius is not fixed.
When I change the cruise speed higher in real time, loiter radius will be changed bigger and not fixed also.
[image]
[image]

Question: The loiter radius set is not working?

hamishwillee, 2018-11-13:
I *THINK* that for fixed wing the radius parameter is being ignored and that the radius is coming from parameter [NAV_LOITER_RAD](https://dev.px4.io/en/advanced/parameter_reference.html#NAV_LOITER_RAD).

The documents *say* that `NAV_LOITER_RAD` is just the default, but on quick scan of the code it looks like it might be always used.

@dagar Is that correct, and if so, why don't we respect the loiter radius from the mission?

goodman110110110, 2018-11-14:
When I change the parameter NAV_LOITER_RAD, the loiter radius is also not changed.
It seems the NAV_LOITER_RAD can not change the radius also. @hamishwillee

hamishwillee, 2018-11-14:
Limited knowledge sorry. Hope maybe @dagar can comment.

closed 2019-02-26 by stale[bot]