PX4/PX4-Autopilot #10839: macOS SITL broken after daemon change

julianoes, 2018-11-14:
SITL doesn't work anymore on macOS.

`git bisect` points to #10766.

I'll look into it.

julianoes, 2018-11-14:
Turns out the daemon client returns immediately and that's what's probably causing the segfault as well as other issues I'm seeing.

julianoes, 2018-11-14:
It looks like this is an issue that we get `poll` for `POLLHUP` too early on macOS. It looks like poll is not equal to poll, it all depends on the OS: https://www.greenend.org.uk/rjk/tech/poll.html

m-ou-se, 2018-11-14:
What's the minimal init script or commands you run to reproduce this?

m-ou-se, 2018-11-14:
I'd be very surprised if POLLHUP comes too early. It coming too late (well, not at all) could make sense, if on Mac the shutdown() doesn't trigger it. But in that case you shouldn't experience much problems, only that the thread doesn't get killed when the client disconnects.

m-ou-se, 2018-11-14:
I was looking at the wrong `shutdown`. It wasn't the `shutdown` of the server thread that caused the problem, but the `shutdown(WR)` of the *client*. But that one is not really needed. Removing it should solve the problem: https://github.com/PX4/Firmware/pull/10846

PR #10846 (2018-11-14): Fix Posix daemon/client on Mac.
Fixes #10839.
closed 2018-11-14 by julianoes