PX4/PX4-Autopilot #10843: Bug: lidarlite accuracy report broken

AuterionWrikeBot, 2018-11-14:
there are reports that the accuracy value report on the lidarlite version !=v3 is not working and giving strange values. To be investigated further.

Apparently it js working on v3, on previous versions and v3 hp it is not

Daniele knows more details

DanielePettenuzzo, 2018-11-14:
There seems to be an issue in the lidar lite driver when running it for the latest Lidar Lite V3HP. Apparently the following register doesn't exist anymore in the HP version:
#define LL40LS_PEAK_STRENGTH_REG  0x0c

This causes the signal quality calculation in the collect() method to fail and the effect is that we always publish a signal_quality=0. I would like to simplify the calculation just for the HP sensor but I was not able to find a way to distinguish a normal V3 from V3HP since also the HW version register doesn't seem to exist anymore on this new sensor.
#define LL40LS_HW_VERSION         0x41

I guess I will just add a parameter to choose between older lidar lites (including v3) and the V3 HP and simplify the quality check just for the V3HP.

DanielePettenuzzo, 2018-11-15:
Sure. I checked all the registers on the lidar lite and there is not id or hw/sw version register. We can have a look today and find a better solution than adding a parameter.

julianoes, 2019-10-14:
I think @mcsauder has fixed this.

closed 2019-10-14 by julianoes