PX4/PX4-Autopilot #10847: QGC: Changing Safety related parameters mid-flight

DonLakeFlyer, 2018-11-14:
I realize this is sort of a support question but it's kind of involved so I figure it would be better here....

Currently in QGC the Safety Setup page is disabled when the vehicle is flying. Sentera has asked that the Data Link Loss settings can be changed mid-flight. This is to support a use case where the vehicle flies out of link range and the user knows it and is working to reposition the antenna.

In order to support this I would like the allow changing of safety values mid-flight which are possible. As far as I can tell from parameter meta data:

Can't be changed in flight:
* Low battery failsafe levels

Can be changed in flight:
* Low battery failsafe action
* All other fields

My question is whether there is the possibility that I implement support for changing these values in flight which will require a ui change only to find the various/all of these are not marked correctly as requiring vehicle reboot to change. And then my time is wasted on the QGC code change.

[image]

MatejFranceskin, 2018-11-14:
I think that all failsafe settings can be changed during the flight. I checked low battery failsafe levels parameters like BAT_CRIT_THR and BAT_LOW_THR in lib/battery.cpp. The values are not stored before the take off but are always read from the parameters - every time the battery status is checked, so the change should be immediately active.

DonLakeFlyer, 2018-11-15:
> I think that all failsafe settings can be changed during the flight.

So then it is a bug that the battery ones are marked as reboot required. Can someone fix that?

DonLakeFlyer, 2018-11-19:
I"m going to allow the full Safety page to display while flying then under the assumption that the various battery fields which are marked incorrectly will be fixed.

MatejFranceskin, 2018-11-19:
In my opinion the doc is wrong - but it would be good if someone with more experience looked at this.

bkueng, 2018-11-20:
I checked a couple:
- SYS_HITL: requires reboot
- COM_LOW_BAT_ACT: no reboot required
- BAT_EMERGEN_THR: param is reloaded, but the warning state is not reset. I.e. if you are already in a low battery state and change the param, it won't reset the warning state.
- NAV_RCL_ACT: no reboot required
- RTL_RETURN_ALT: no reboot required
- COM_DISARM_LAND: no reboot required

The rest should not require a reboot, and can be changed in-flight.

hamishwillee, 2018-11-20:
@bkueng The metadata correct for all of those right? https://github.com/PX4/Firmware/issues/10847#issuecomment-440188928

bkueng, 2018-11-20:
Yes.

closed 2019-03-04 by stale[bot]