PX4/PX4-Autopilot #10850: Quadcopter losing altitude in mission mode with Pixhawk 4

neom89, 2018-11-15:
I have a quadcopter with **Pixhawk 4** on board containing **1.8.1** version of PX4 firmware. While doing a mission it loses altitude but when I put it on the loiter mode the altitude is maintained perfectly. 
I analyzed the logs but have not been able to pin point the cause for the same. 

Here is the log for loiter mode:
https://review.px4.io/plot_app?log=b3ad1be5-4042-4d18-8107-edbb41db6bc8

Here are a couple of logs of the mission mode:
https://review.px4.io/plot_app?log=455b0c9f-4609-4dfe-9d10-78a05c68e747
https://review.px4.io/plot_app?log=d7e42330-5ac4-454d-a990-750ba5047614
https://review.px4.io/plot_app?log=67d9d40d-796c-4b9d-b37d-0fcd2e9e29a8

Crowdedlight, 2018-12-30:
We observed the same behaviour, but that was when we tried to make the drone fly faster than it could with the setup while keeping the altitude. We also had to properly calibrate the hover-mid-throttle. That together with lowering the horizontal speed helped us keep the altitude when flying mission.

neom89, 2018-12-31:
@Crowdedlight Are you using Pixhawk 4?

Crowdedlight, 2019-01-02:
@neom89 ah sorry. Must have missed that. No this was Pixhawk 2.1.

closed 2019-07-08 by stale[bot]

julianoes, 2019-07-15:
@bresch is there tuning that we should document for better altitude hold in forward flight?