PX4/PX4-Autopilot #10852: Getting 'Preflight Fail: Accel Range, hold still on arming' in Gazebo SITL with custom drone

varunvp, 2018-11-15:
**Describe the bug**
I had to run the PX4 SITL with our own custom drone instead of the Iris drone. I made a copy of the rotors_description file, and I replaced the necessary parameters and models. I modified the posix_sitl.launch file to the following

```
<?xml version="1.0"?>
<launch>
    <!-- Posix SITL environment launch script -->
    <!-- launches PX4 SITL, Gazebo environment, and spawns vehicle -->
    <!-- vehicle pose -->
    <arg name="x" default="0"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>
    <!-- vehicle model and world -->
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="raven"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    <arg name="ID" default="1"/>
    <arg name="mavlink_udp_port" default="14560"/>
    <env name="PX4_SIM_MODEL" value="iris" />      <----Had to set this manually to iris
    <env name="PX4_ESTIMATOR" value="$(arg est)" />

    <!-- gazebo configs -->
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    <arg name="respawn_gazebo" default="false"/>
    <!-- PX4 configs -->
    <arg name="interactive" default="true"/>
    <!-- PX4 SITL -->
    <arg unless="$(arg interactive)" name="px4_command_arg1" value="-d"/>
    <arg     if="$(arg interactive)" name="px4_command_arg1" value=""/>
    <node name="sitl" pkg="px4" type="px4" output="screen"
        args="$(find px4)/ROMFS/px4fmu_common -s etc/init.d-posix/rcS $(arg px4_command_arg1)" required="true"/>

    <!--I wanted to use the URDF model, instead of SDF, for ease of adding sensors, so I copied the following from another launch file-->
    <!-- generate urdf vehicle model -->
    <arg name="cmd" default="$(find xacro)/xacro $(find px4)/Tools/sitl_gazebo/models/raven/urdf/$(arg vehicle)_base.xacro raven_dir:=$(find px4)/Tools/sitl_gazebo/models/raven mavlink_udp_port:=$(arg mavlink_udp_port) --inorder"/>    
    <param command="$(arg cmd)" name="raven"/>
    <!-- Gazebo sim -->
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
        <arg name="respawn_gazebo" value="$(arg respawn_gazebo)"/>
    </include>
    <!-- Changes to accomodate for launching a URDF instead of SDF -->
    <!-- gazebo model -->
    <node name="$(anon vehicle_spawn)_$(arg ID)_spawn" output="screen" pkg="gazebo_ros" type="spawn_model" args="-urdf -param raven -model $(arg vehicle)_$(arg ID) -package_to_model -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>
</launch>
```
When I launch the file after setting the necessary environmental variables, the simulator starts up. When I type commander takeoff, I get the following error
```
WARN  [commander_tests] Preflight Fail: Accel Range, hold still on arming
ERROR [commander] arming failed
Command 'commander' failed, returned 1.
```

I typed sensors status, and I get the following
```
INFO  [sensors] gyro status:
INFO  [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 100, state: OK
INFO  [ecl/validation] 	val:  -0.0095, lp:  -0.0003 mean dev:  -0.0000 RMS:   0.0083 conf:   1.0000
INFO  [ecl/validation] 	val:   0.0055, lp:  -0.0066 mean dev:  -0.0000 RMS:   0.0157 conf:   1.0000
INFO  [ecl/validation] 	val:   0.0003, lp:   0.0005 mean dev:   0.0000 RMS:   0.0060 conf:   1.0000
INFO  [sensors] accel status:
INFO  [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 100, state: OK
INFO  [ecl/validation] 	val:  -0.9545, lp:  -0.2702 mean dev:  -0.0005 RMS:   0.2567 conf:   1.0000
INFO  [ecl/validation] 	val:   0.3974, lp:   0.1784 mean dev:  -0.0016 RMS:   0.1082 conf:   1.0000
INFO  [ecl/validation] 	val: -18.9993, lp: -15.4979 mean dev:   0.0088 RMS:   7.4416 conf:   1.0000
INFO  [ecl/validation] sensor #1, prio: 75, state: OK
INFO  [ecl/validation] 	val:  -0.9549, lp:  -0.2669 mean dev:  -0.0000 RMS:   0.4700 conf:   1.0000
INFO  [ecl/validation] 	val:   0.3954, lp:   0.1784 mean dev:  -0.0020 RMS:   0.1286 conf:   1.0000
INFO  [ecl/validation] 	val: -18.9993, lp: -15.4237 mean dev:   0.0107 RMS:   7.7129 conf:   1.0000
INFO  [sensors] mag status:
INFO  [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 50, state: OK
INFO  [ecl/validation] 	val:   0.0159, lp:   0.0065 mean dev:   0.0000 RMS:   0.0101 conf:   1.0000
INFO  [ecl/validation] 	val:  -0.2181, lp:  -0.2149 mean dev:  -0.0001 RMS:   0.0100 conf:   1.0000
INFO  [ecl/validation] 	val:  -0.4248, lp:  -0.4278 mean dev:  -0.0001 RMS:   0.0102 conf:   1.0000
INFO  [sensors] baro status:
INFO  [ecl/validation] validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  [ecl/validation] sensor #0, prio: 75, state: OK
INFO  [ecl/validation] 	val: 955.9515, lp: 955.9492 mean dev:   0.0001 RMS:   0.0100 conf:   1.0000
INFO  [ecl/validation] 	val:  32.0000, lp:  32.0000 mean dev:   0.0000 RMS:   0.0000 conf:   1.0000
INFO  [ecl/validation] 	val:   0.0000, lp:   0.0000 mean dev:   0.0000 RMS:   0.0000 conf:   1.0000
INFO  [sensors] Temperature Compensation:
INFO  [sensors]  gyro: enabled: 0
INFO  [sensors]  accel: enabled: 0
INFO  [sensors]  baro: enabled: 0
INFO  [sensors] Airspeed status:
INFO  [ecl/validation] 	no data
```
You can see that the accel sensor is reporting a status of -18 in the z axis, which is abnormal. I went ahead and checked in Gazebo the forces being applied on the drone.

This is for the normal iris drone
[image]

And for our custom drone
[image]

You can see a drastic difference in the body_2_wrench/force/z value. For the normal one it is close to zero, and for the other one it is close to infinity! Is this the source of the error?

**To Reproduce**
Steps to reproduce the behavior:
1. Unzip the following file to Firmware/Tools/sitl_gazebo/models, make sure folder is named as raven
[raven.tar.gz](https://github.com/PX4/Firmware/files/2583770/raven.tar.gz)
2. In mavros_posix_sitl.launch file, change the include in line 30 to point to the following launch file, which should be placed in the launch folder. Comment line 39 and line 38.
[raven_sitl.txt](https://github.com/PX4/Firmware/files/2583772/raven_sitl.txt)
3. Launch mavros_posix_sitl.launch
4. You can follow what I've described above to see the errors.

closed 2019-02-13 by TSC21

CalvinKroese, 2019-03-28:
Did you ever find the solution to this?

varunvp, 2019-03-29:
I must have, I don't remember. Basically just take a look at the drone from up close. If it's vibrating, then you get this error. Usually happens when there is something wrong with the drone model, or it's placed in an awkward position.

robin-shaun, 2021-02-26:
Arming again may help.
[image]

haozi6, 2022-05-07:
@robin-shaun did you solve this? I get the same problem!老师您好，请问你们解决了这个问题嘛，我们初步推断是模型抖动的原因，但是不知道如何解决。