PX4/PX4-Autopilot #10876: Cannot compile px4fmu-v2_default

matteoscanavino, 2018-11-20:
Hello, 
I'm in trouble whith gcc compiler while trying to build px4 code.

As reported in https://dev.px4.io/en/setup/building_px4.html, I must have a supported GCC version to build make px4fmu-v2_default. I've followed the instructions in https://dev.px4.io/en/setup/dev_env_linux_ubuntu.html#nuttx-based-hardware, I've executed the script GCC 7-2017-q4 and reboot my computer. However, I still have arm-none-eabi-gcc 4.9.3 when I check the version (instead of version 7.2.1 as reported in the docs).

I've noticed that when executing sudo apt-get remove gcc-arm-embedded I got the following:

Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package gcc-arm-embedded

I'm working with Ubuntu 16.04 LTS.
If I try to compile px4fmu-v2_default with my GCC, the process fails and the following error is give: 

[644/646] Linking CXX executable nuttx_px4fmu-v2_default.elf
FAILED: : && /opt/gcc-arm-none-eabi-4_9-2015q3/bin/arm-none-eabi-g++   -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Wextra -Werror -Warray-bounds -Wdisabled-optimization -Wdouble-promotion -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-implicit-fallthrough -Wno-missing-field-initializers -Wno-missing-include-dirs -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wno-overloaded-virtual -Wreorder -Wno-format-truncation -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -fno-math-errno -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG  -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -nodefaultlibs -nostdlib -Wl,--warn-common,--gc-sections platforms/nuttx/CMakeFiles/nuttx_px4fmu-v2_default.elf.dir/__/__/src/platforms/empty.c.obj  -o nuttx_px4fmu-v2_default.elf -L/home/matteo/src/Firmware/build/nuttx_px4fmu-v2_default/external/Install/lib -nodefaultlibs -nostdlib -fno-exceptions -fno-rtti -Wl,--script=/home/matteo/src/Firmware/build/nuttx_px4fmu-v2_default/NuttX/nuttx/configs/px4fmu-v2/scripts/ld.script -Wl,-Map=nuttx_px4fmu-v2_default.map -Wl,--warn-common -Wl,--gc-sections -Wl,--start-group NuttX/apps/libapps.a NuttX/nuttx/arch/arm/src/libarch.a NuttX/nuttx/binfmt/libbinfmt.a NuttX/nuttx/libc/libc.a NuttX/nuttx/configs/libconfigs.a NuttX/nuttx/libxx/libcxx.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sched/libsched.a -Wl,--end-group -lm src/modules/uORB/libmodules__uORB.a src/drivers/barometer/ms5611/libdrivers__ms5611.a src/drivers/differential_pressure/ms4525/libdrivers__ms4525_airspeed.a src/drivers/distance_sensor/ll40ls/libdrivers__ll40ls.a src/drivers/distance_sensor/sf0x/libdrivers__sf0x.a src/drivers/gps/libdrivers__gps.a src/drivers/imu/l3gd20/libdrivers__l3gd20.a src/drivers/imu/lsm303d/libdrivers__lsm303d.a src/drivers/imu/mpu6000/libdrivers__mpu6000.a src/drivers/imu/mpu9250/libdrivers__mpu9250.a src/drivers/magnetometer/hmc5883/libdrivers__hmc5883.a src/drivers/px4flow/libdrivers__px4flow.a src/drivers/px4fmu/libdrivers__px4fmu.a src/drivers/px4io/libdrivers__px4io.a src/drivers/rgbled/libdrivers__rgbled.a src/drivers/stm32/adc/libdrivers__adc.a src/drivers/stm32/tone_alarm/libdrivers__tone_alarm.a src/drivers/vmount/libdrivers__vmount.a src/modules/sensors/libmodules__sensors.a src/systemcmds/bl_update/libsystemcmds__bl_update.a src/systemcmds/hardfault_log/libsystemcmds__hardfault_log.a src/systemcmds/mixer/libsystemcmds__mixer.a src/systemcmds/mtd/libsystemcmds__mtd.a src/systemcmds/param/libsystemcmds__param.a src/systemcmds/perf/libsystemcmds__perf.a src/systemcmds/pwm/libsystemcmds__pwm.a src/systemcmds/reboot/libsystemcmds__reboot.a src/systemcmds/top/libsystemcmds__top.a src/systemcmds/tune_control/libsystemcmds__tune_control.a src/systemcmds/ver/libsystemcmds__ver.a src/modules/camera_feedback/libmodules__camera_feedback.a src/modules/commander/libmodules__commander.a src/modules/events/libmodules__events.a src/modules/land_detector/libmodules__land_detector.a src/modules/load_mon/libmodules__load_mon.a src/modules/mavlink/libmodules__mavlink.a src/modules/navigator/libmodules__navigator.a src/modules/ekf2/libmodules__ekf2.a src/modules/fw_att_control/libmodules__fw_att_control.a src/modules/fw_pos_control_l1/libmodules__fw_pos_control_l1.a src/modules/mc_att_control/libmodules__mc_att_control.a src/modules/mc_pos_control/libmodules__mc_pos_control.a src/modules/vtol_att_control/libmodules__vtol_att_control.a src/modules/logger/libmodules__logger.a src/modules/dataman/libmodules__dataman.a src/modules/uORB/libmodules__uORB.a ROMFS/libromfs.a src/lib/drivers/airspeed/libdrivers__airspeed.a src/lib/led/libled.a src/lib/battery/libbattery.a src/lib/ecl/validation/libecl_validation.a src/lib/mixer/libmixer.a src/lib/tunes/libtunes.a src/lib/DriverFramework/framework/src/libdf_driver_framework.a src/lib/airspeed/libairspeed.a src/lib/ecl/EKF/libecl_EKF.a src/lib/ecl/geo_lookup/libecl_geo_lookup.a src/lib/ecl/attitude_fw/libecl_attitude_fw.a src/lib/landing_slope/liblanding_slope.a src/lib/ecl/l1/libecl_l1.a src/lib/ecl/tecs/libecl_tecs.a src/modules/fw_pos_control_l1/launchdetection/liblaunchdetection.a src/modules/fw_pos_control_l1/runway_takeoff/librunway_takeoff.a src/lib/circuit_breaker/libcircuit_breaker.a src/lib/conversion/libconversion.a src/lib/mathlib/libmathlib.a src/lib/ecl/geo/libecl_geo.a src/lib/controllib/libcontrollib.a src/lib/FlightTasks/libFlightTasks.a src/lib/FlightTasks/tasks/ManualAltitudeSmooth/libFlightTaskManualAltitudeSmooth.a src/lib/FlightTasks/tasks/ManualPositionSmooth/libFlightTaskManualPositionSmooth.a src/lib/FlightTasks/tasks/ManualPositionSmoothVel/libFlightTaskManualPositionSmoothVel.a src/lib/FlightTasks/tasks/ManualPosition/libFlightTaskManualPosition.a src/lib/FlightTasks/tasks/ManualAltitude/libFlightTaskManualAltitude.a src/lib/FlightTasks/tasks/ManualStabilized/libFlightTaskManualStabilized.a src/lib/FlightTasks/tasks/Manual/libFlightTaskManual.a src/lib/FlightTasks/tasks/AutoLine/libFlightTaskAutoLine.a src/lib/FlightTasks/tasks/AutoMapper/libFlightTaskAutoMapper.a src/lib/FlightTasks/tasks/AutoLineSmoothVel/libFlightTaskAutoLineSmoothVel.a src/lib/FlightTasks/tasks/Utility/libFlightTaskUtility.a src/lib/FlightTasks/tasks/AutoMapper2/libFlightTaskAutoMapper2.a src/lib/FlightTasks/tasks/AutoFollowMe/libFlightTaskAutoFollowMe.a src/lib/FlightTasks/tasks/Auto/libFlightTaskAuto.a src/lib/FlightTasks/tasks/Offboard/libFlightTaskOffboard.a src/lib/FlightTasks/tasks/Failsafe/libFlightTaskFailsafe.a src/lib/FlightTasks/tasks/Transition/libFlightTaskTransition.a src/lib/FlightTasks/tasks/FlightTask/libFlightTask.a src/lib/WeatherVane/libWeatherVane.a src/lib/pwm_limit/libpwm_limit.a src/lib/version/libversion.a src/platforms/common/libplatforms__common.a platforms/nuttx/src/px4_layer/libpx4_layer.a NuttX/nuttx/arch/arm/src/libarch.a src/drivers/boards/px4fmu-v2/libdrivers_board.a src/drivers/stm32/libdrivers_arch.a src/drivers/boards/common/libdrivers_boards_common.a src/drivers/boards/common/stm32/libdrivers_boards_common_arch.a src/modules/systemlib/libsystemlib.a src/lib/drivers/device/libdrivers__device.a src/lib/drivers/led/libdrivers__led.a src/lib/parameters/libparameters.a src/lib/cdev/libcdev.a src/lib/parameters/flashparams/libflashparams.a NuttX/nuttx/arch/arm/src/libarch.a src/drivers/boards/px4fmu-v2/libdrivers_board.a src/drivers/stm32/libdrivers_arch.a src/drivers/boards/common/libdrivers_boards_common.a src/drivers/boards/common/stm32/libdrivers_boards_common_arch.a src/modules/systemlib/libsystemlib.a src/lib/drivers/device/libdrivers__device.a src/lib/drivers/led/libdrivers__led.a src/lib/parameters/libparameters.a src/lib/cdev/libcdev.a src/lib/parameters/flashparams/libflashparams.a NuttX/apps/libapps.a src/lib/perf/libperf.a src/lib/parameters/tinybson/libtinybson.a NuttX/nuttx/libxx/libcxx.a NuttX/nuttx/libc/libc.a NuttX/nuttx/drivers/libdrivers.a NuttX/nuttx/fs/libfs.a NuttX/nuttx/mm/libmm.a NuttX/nuttx/sched/libsched.a -lm -lgcc msg/libuorb_msgs.a && :
[... cut ...]
/opt/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: address 0x8105fa4 of nuttx_px4fmu-v2_default.elf section `__param' is not within region `flash'
/opt/gcc-arm-none-eabi-4_9-2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld: region `flash' overflowed by 27989 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Makefile:176: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1

Any suggestions on how to fix it?
Thank-you in advance

hamishwillee, 2018-11-20:
The build error is definitely because you're not using the right GCC compiler - as you said.  YOu'll need to uninstall arm-none-eabi-gcc 4.9.3 or do whatever it takes to point to GCC 7-2017-q4 - 

If you followed this fully you will have done this before installing the new toolchain:

- Remove any old versions of the arm-none-eabi toolchain.
```
sudo apt-get remove gcc-arm-none-eabi gdb-arm-none-eabi binutils-arm-none-eabi gcc-arm-embedded
sudo add-apt-repository --remove ppa:team-gcc-arm-embedded/ppa
```

If you were installing on a clean system I would suggest just running the [standard build script](https://dev.px4.io/en/setup/dev_env_linux.html#pixhawknuttx-and-jmavsim) - it does the same things as the instructions but you don't have to think about it.

matteoscanavino, 2018-11-20:
@hamishwillee 
Thankyou. When I run sudo apt-get remove gcc-arm-none-eabi gdb-arm-none-eabi binutils-arm-none-eabi gcc-arm-embedded I get this message: 

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package gcc-arm-embedded

Furthermore arm-none-eabi-gcc --version gives still arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]

matteoscanavino, 2018-11-20:
I solved the problem removing manually the folder gcc-arm-none-eabi-4_9-2015q3: for some strange reason, when I tried to compile GCC4.9.3 was used by default. By removing the folder, GCC7-2017-q4 was used and I was able to build the code.

closed 2018-11-20 by matteoscanavino