PX4/PX4-Autopilot #10878: Simulator not using distance_sensor

tuloski, 2018-11-20:
I'm running the simulator with gazebo and I have simulated a laser sensor.
With MAVros I send the mavlink DISTANCE_SENSOR to the autopilot and I see correctly the message on qgroundcontrol->mavlink inspector, but the quote from the lidar is not used at all.
I'm using the parameters:
* MPC_ALT_MODE Terrain Follow
* EKF2_RNG_AID Enabled
* EKF2_HGT_MODE Range sensor

I'm wondering if it is a bug in the simulator only, or a problem with the px4/firmware which doesn't accept external DISTANCE_SENSOR messages (but it seems external messages are read and published on uorb https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L818-L845)?

mrivi, 2018-11-20:
@tuloski DISTANCE_SENSOR plugin is blacklisted by default in mavros [here](https://github.com/mavlink/mavros/blob/de9f39a719b091b8448214a17d27b3b1c415d0dc/mavros/launch/px4_pluginlists.yaml)

If you remove it, it should work.

tuloski, 2018-11-20:
I removed it, and as I said I see the message arriving in qground mavlink inspector, so it's arriving to the autopilot.

mrivi, 2018-11-20:
@tuloski sorry, I misread.

tuloski, 2018-11-20:
Can it be that the timestamp of the autopilot and the distance_sensor are different, since distance_sensor timestamp comes from ROS.
How can I check in the EKF which sensors are being used for the estimation? Maybe somewhere it's written that the lidar data was dropped because of some motivation...

closed 2019-03-04 by stale[bot]