PX4/PX4-Autopilot #10881: Strange drift on velocity tracking

limhyon, 2018-11-20:
https://logs.px4.io/plot_app?log=06cbb7ce-198a-45ff-9e9f-557143f20b93

Here is the log.
Around 2:00, there is a significant divergence on velocity tracking, but until then, it is fine.

P.S.
https://logs.px4.io/plot_app?log=a14f4203-1ad1-4a58-a9ee-cfde69bf0d9a
another log too.

MaEtUgR, 2018-11-20:
@limhyon Do you have any more detailed information? Like what did you see happening which wasn't expected? Did it drift off like a flyaway or just fly wobbly?

MaEtUgR, 2018-11-20:
I just saw that at 1:58 it quickly detects ground contact which could be either the cause or a consequence of the altitude divergence.

MaEtUgR, 2018-11-20:
> I was testing GPS hovering
> and it maintained position quite well
> but suddenly, it moves in one direction straight
> so that I changed to stabilize mode

dakejahl, 2018-11-21:
It's because of **EKF_RNG_AID**. I don't understand the exact mechanism of failure, but things go severely wrong when you toggle between baro and range finder. 
[image]

LorenzMeier, 2018-11-21:
@priseborough FYI

MaEtUgR, 2020-02-19:
@limhyon Is this problem still present? There were a lot of changes and fixes, I wouldn't know which one exactly is related.

bresch, 2020-05-20:
There have been a lot of bug fixes and refactoring in the switch between different sources of height fusion. I didn't see this issue anymore, closing.

closed 2020-05-20 by bresch