PX4/PX4-Autopilot #10894: bl_update support for F7

davids5, 2018-11-21:
IIRC F7 has bl_update disabled. It needs to be ported to F7 and vetted. 

The F7 (FMUv5) bootloader enhancements for high speed uart is not deployed.  This is a blocker.
Also (FMUv5) bootloader used the wrong status LED signaling.

dagar, 2018-11-21:
Currently we're including the px4fmuv3 bootloader firmware to make it easy to upgrade pixhawk derivatives that shipped with the old bootloader. Once bl_update is vetted on F7 I can extend the build system to bake in the latest bootloader for every board.

davids5, 2019-02-13:
@dagar please make it so. https://github.com/PX4/Firmware/pull/11394

mhkabir, 2019-07-14:
@julianoes this is in master.

davids5, 2019-07-14:
@mhkabir both parts?

mhkabir, 2019-07-14:
Oops, bootloader binary isn't baked in yet - sorry!

dagar, 2019-07-14:
My plan was to get the bootloader structure (naming, etc) in sync first so it's a little saner to manage. https://github.com/PX4/Bootloader/pull/137