PX4/PX4-Autopilot #10899: HITL takeoff issue

matteoscanavino, 2018-11-22:
Hello,
I'd like to perform HITL simulation. I've followed the tutorial reported in https://dev.px4.io/en/simulation/hitl.html and I've set my Gazebo simulation environment as described. 
I've modified the iris.sdf file as requested and calibrated my RC. 
When I launch Gazebo in HITL mode using gazebo Tools/sitl_gazebo/worlds/iris.world, I can arm the copter from my RC but unfortunately the iris doesn't take off at all, even if I set the throttle at max. 
I'm working with ROS Kinetic, Gazebo 7 and px4 firmware v1.8.0-1772-g919a508. 

Does anyone experience the same issue?
Thankyou in advance

bkueng, 2018-11-23:
Hi
Which board are you using?

S259420, 2018-11-23:
I'm using a PixHawk 1. I've compiled and uploaded the firmware with px4fmu-v2_default upload

matteoscanavino, 2018-11-23:
@bkueng I'm using a PixHawk 1. I've compiled and uploaded the firmware with px4fmu-v2_default upload. The throttle command is received by the board as I can see it in QGround... I don't have any idea how to fix it

bkueng, 2018-11-26:
Please update the bootloader (https://docs.px4.io/en/advanced_config/bootloader_update.html), and flash the v3 target if possible. v2 does not include the necessary HIL driver because of limited flash space.

closed 2019-03-10 by stale[bot]