PX4/PX4-Autopilot #10902: Build PX4 on Windows using Eclipse/MSVS2017 IDE

evgenee, 2018-11-23:
Hi,
I am very interested to setup a  PX4 build environment on Windows using Eclipse or MS VS 2017 IDE. 
I tried to use last PX4 toolchain 0.4 for building on Eclipse but there is no preconfigured Eclipse as it was in previous versions of pixhawk_toolchain and with new versions I run into problems.

I am looking for some help/suggestions with this.
Thanks

dagar, 2018-11-24:
For eclipse the integration isn't perfect, but you can have an eclipse makefile project that simply calls out to the real build system.

To start I'd try copying `eclipse.cproject -> .cproject` and `eclipse.project -> .project`, then in eclipse import project.

hamishwillee, 2018-11-25:
Hi @evgenee 

I *think* there is a preconfigured eclipse, but it is incorrectly documented. Specifically, the [usage instructions](https://dev.px4.io/en/setup/dev_env_windows_cygwin.html#usage_instructions) state that the batch files are in C:/PX4, when actually they are copied into start menu (ie type PX4 into start menu, and you should see "PX4 Exclipse" option.

Can you try that and let me know what more we should say?

MaEtUgR, 2018-11-26:
@evgenee Hey, nice I catched your interest. I can send you all information needed to build within eclipse. It's actually quite simple. I removed eclipse from the PX4 toolchain because:
- I wanted to shrink download and installation size
- I assumed everyone uses his own editor and wasn't sure about the number of eclipse users
- I myself started using [Visual Studio Code](https://code.visualstudio.com/) and wanted to cut my efforts to serve and test multiple editors

Here's a short summary of how I used eclipse in previous toolchain versions:
1. download latest or prefered eclipse version with C++ (CDT) support: https://www.eclipse.org/cdt/
1. unpack it to the desired location (for me it was `C:\PX4\toolchain\eclipse\`)
1. Create a batch script that calls the setup-environment file (name and location changed from 0.4 to 0.5)
https://github.com/PX4/windows-toolchain/blob/master/run-console.bat#L1-L4
and calls eclipse (instead of bash for the console)
https://github.com/PX4/windows-toolchain/blob/master/run-console.bat#L14
1. after successfully starting eclipse with your batch script it runs within the cygwin environment with the PATH of the toolchain and hence you can import the Firmware folder as Makefile project `Import ->C/C++ -> Makefile Project With Existing Code` (don't remember the names exactly), add your make targets and start developing.

AFAIK "MS VS 2017 IDE" is designed/optimized to build on the Microsoft Visual C++ compiler which we do not support so far. That's why I would go with a general purpose IDE like eclipse or VS code. But if you want to have a try I'm happy to help with arising questions and if you could contribute documentation for what you achieve.

evgenee, 2018-11-26:
Hi,
@MaEtUgR thank you very much!
I will try the steps with Eclipse and report here about my progress.

Anyway, I really want to set up the environment on MSVS2017 community edition. It is very nice IDE that currently has a support for a lot of new features (all features that VS Code and more). See these links for example:

https://blogs.msdn.microsoft.com/vcblog/2017/04/11/linux-development-with-c-in-visual-studio/

https://blogs.msdn.microsoft.com/vcblog/2017/07/19/using-mingw-and-cygwin-with-visual-cpp-and-open-folder/

https://blogs.msdn.microsoft.com/vcblog/2017/10/23/arm-gcc-cross-compilation-in-visual-studio/

https://youtu.be/qWAPyGN56Pk

https://docs.microsoft.com/en-us/cpp/ide/cmake-tools-for-visual-cpp?view=vs-2017

Sure, I will be very glad to have your help with this adventure.

Thanks again!

MaEtUgR, 2018-11-26:
> Anyway, I really want to set up the environment on MSVS2017 community edition.
> Sure, I will be very glad to have your help with this adventure.

Sounds good 👍 

Comment on the links:
1. Linux development is cross compilation, it's not for a Windows target system. I just saw it works with the Linux subsystem on Windows. My main issue with this system is graphics (2D/3D) and connection to hardware devices support.
2. Sounds interesting, so it seems they have support for the most common posix environments on Windows now.
3. Exceptionally interesting to me. The example they have is based on mbed (the ARM arduino-like platform) which I used to program my drone before joining the PX4 community (see https://github.com/MaEtUgR/FlyBed).
4. Have to watch later, but it seems to say something similar to 2.
5. That one I knew already. But it would be not as useful to us if it would still not support any other compiler like gcc.

Btw this might be also interesting to you https://github.com/PX4/Firmware/pull/10911 vscode configuration with cmake and debugging, didn't check if any of it works on windows. We might need to start the IDE in a similar fashion I described above for eclipse to make it work.

> The option of calling toolchain preconfigured eclipse from windows command line does't exist with latest toolchains..(ver. 0.3 and 0.4)

We can readd it on https://github.com/PX4/windows-toolchain. I just didn't want to have something in there that I currently cannot test and support.

closed 2019-03-10 by stale[bot]

hkuyam008, 2019-04-03:
I understand this ticket is stale and has been closed automatically, but I wanted to record my observations. 

As stated above in the discussion thread the run-eclipse.bat file has been removed from the installer MSI due to various reasons, but mostly due to size being an issue.

So I tried the following suggested above -
download latest or prefered eclipse version with C++ (CDT) support: https://www.eclipse.org/cdt/
unpack it to the desired location (for me it was C:\PX4\toolchain\eclipse\)
Create a batch script that calls the setup-environment file (name and location changed from 0.4 to 0.5)
https://github.com/PX4/windows-toolchain/blob/master/run-console.bat#L1-L4
and calls eclipse (instead of bash for the console)
https://github.com/PX4/windows-toolchain/blob/master/run-console.bat#L14
after successfully starting eclipse with your batch script it runs within the cygwin environment with the PATH of the toolchain and hence you can import the Firmware folder as Makefile project Import ->C/C++ -> Makefile Project With Existing Code (don't remember the names exactly), add your make targets and start developing.

But the jmavsim console opens up and shuts down as I noted in another ticket.

MaEtUgR, 2019-04-05:
> But the jmavsim console opens up and shuts down as I noted in another ticket.

ok, so we can leave it closed and follow up in the other issue. I need to try it myself to check what's going on all other input I already wrote.

sani1486, 2019-07-23:
@MaEtUgR I am very eager to know how you configure VS code for PX4 can you please give me more insight? 
Thanks