PX4/PX4-Autopilot #10906: Improved Fixed Wing Loss of GPS (global position) failsafe behavior when landing

Antiheavy, 2018-11-24:
The Fixed Wing loss of GPS failsafe triggers the "open loop loiter" which is great when at mission altitude.

However, in the special case of GPS loss when on a landing approach we do not want the aircraft going into a bank. In this case, the preferred behavior is to continue straight ahead on a dumb landing path (cut throttle, control airspeed/pitch. hold current body yaw angle). Regain of GPS should pick-up where it left off on the mission landing and get back onto the guided landing approach.

The simpliest way to improve this would be to override NAV_GPSF_R and force it to 0 degrees bank angle if entering GPS Failure Navigation from a Landing approach.

almaaro, 2019-01-12:
One solution would be: In navigator_main.cpp in 
`case vehicle_status_s::NAVIGATION_STATE_AUTO_LANDENGFAIL:` 
there would be a check to see if the current waypoint type is LAND to avoid entering the open loop loiter. Also, in FixedwingPositionControl.cpp there would be a check to see if the GPS is in failsafe to enable specific landing control.

Also, some other thoughts arise regarding the whole GPS failsafe.
- As far as I've understood it, there is no altitude control for GPS control loss (pitch, roll and throttle are controlled by NAV_GPSF_x). So could the whole open loop loiter just be moved to fw position controller to control the altitude normally and to just output a static roll angle?
    - If the failsafe gets moved to the fw pos controller, I think it would be justified to apply a minimum altitude for the loiter (the plane would climb first and then start the loiter).
    - If landing (or after just passing LAND_START), the plane would just climb first and then start the loiter, like a go-around? After regaining GPS, the plane would continue from LAND_START if it exists, otherwise from the wpt before LAND.
        - Or would it be safer to just bring the plane down without GPS if it has already passed LAND. This would be better if the GPS position is never coming back (hardware fault).

JohnSnowball, 2019-02-16:
I got a question, when you suffer gps loss during mission, will your fixed wing goes to open-loop loiter every time? or just go landing slope?

Antiheavy, 2019-02-16:
> I got a question, when you suffer gps loss during mission, will your fixed wing goes to open-loop loiter every time? or just go landing slope?

@JohnSnowball I am noticing that the documentation on this isn't very good:
https://docs.px4.io/en/config/safety.html
https://docs.px4.io/en/advanced_config/parameter_reference.html#gps-failure-navigation
Multiple terms are used "loss of position", "loss of global position", and "loss of GPS".  They all have slightly different meanings are are not used consistently. ( @hamishwillee maybe something for a separate discussion)

Here is more information specific to fixed wing:
When you lose GPS (i'm talking losing satellite based position lock here) the system doesn't do anything at first.  The EKF2 continues to estimate its global position using the other sensors and dead reckoning.  The EKF2 can do a pretty good job for several seconds or even up to several minutes on a fixed wing that uses an airspeed sensor and has both EKF2_ARSP_THR and EKF2_FUSE_BETA enabled.  Once the EKF2's reported estimates for position and velocity errors exceed the position failsafe thresholds in Commander (i.e. COM_POS_FS_xx...) then the "GPS Failure Navigation" kicks in.

GPS Failure Navigation is NOT a landing glide slope.  It just sets the aircraft to fixed pitch, bank, and throttle settings.  The default parameters for NAV_GPSF_xx result in an open loop loiter down to the ground on most fixed wing aircraft.  If the reported estimates for position and velocity errors come back within the limits then the vehicle resumes whatever mode it was in before (e.g. back into Mission mode and climbs and turns back to the waypoint it was tracking).

The feature request in this issue is to override NAV_GPSF_R and set to 0 if entering GPS Failure Navigation from a "Land Mode" (for fixed wings).

Antiheavy, 2019-06-25:
@kjkinney let’s look at this

Antiheavy, 2019-09-23:
I think we may have an inprovement for this coming soon. Keep this issue open for that.