PX4/PX4-Autopilot #10915: About fly mission

roseyangyu, 2018-11-26:
I have set a mission.But I have a problem.I hope vechicle to hold 3 secs after take a photo.But the command is hold frist,than take a photo.How do I set up to give priority to taking pictures?
[image]

like this?I want the hold-command to be after take photo
[image]

hamishwillee, 2018-11-26:
@DonLakeFlyer Is this a QGC issue or a Firmware issue?

hamishwillee, 2018-11-26:
@roseyangyu  This is a support question, not an issue. If Don's answer is not sufficient for you, then please expand on the question in our add to our [forum/chat channels](https://docs.px4.io/en/#support) (Github is for bugs, not questions).

closed 2018-11-26 by hamishwillee

roseyangyu, 2018-11-27:
thank you for your help