PX4/PX4-Autopilot #10923: Extension field not initialized in HOME_POSITION message

garfieldG, 2018-11-27:
Hi,

I've noticed that when receiving HOME_POSITION from fcc the extension field of the message (time_usec) is sometimes filled with garbage values.

I think the problem is that the field is not initialized in mavlink_messages as the other non extension fields of the message :
https://github.com/PX4/Firmware/blob/48df19c8dfa145a16278b48e2aff0a4eda74feda/src/modules/mavlink/mavlink_messages.cpp#L2686

mhkabir, 2018-11-27:
Can you send a PR? Thanks!

BazookaJoe1900, 2018-12-05:
@garfieldG , @mhkabir 
I went over all messages (with extentions in praticular), and found more issues:
1)
https://github.com/PX4/Firmware/blob/311c0941c995ccc996df2494b4893007af49e6a5/src/modules/mavlink/mavlink_messages.cpp#L3425 
Doesn't init chan9_raw-chan18_raw, and should they be 0?

2)
https://github.com/PX4/Firmware/blob/311c0941c995ccc996df2494b4893007af49e6a5/src/modules/mavlink/mavlink_mission.cpp#L371

Here I am not sure. does 'mission_type' isn't being set inside the function and how to fix.

dagar, 2018-12-27:
Fixed in https://github.com/PX4/Firmware/pull/11105

closed 2018-12-27 by dagar