PX4/PX4-Autopilot #10929: AUTO.LAND mode triggering setpoint to (0,0,0)

Seanmatthews, 2018-11-27:
**Describe the bug**
I am using MAVROS to control PX4 by sending setpoints in OFFBOARD mode. During my commanded flight, I issue a submode change to AUTO.LAND (see "PX4 native flight stack" section [here](http://wiki.ros.org/mavros/CustomModes)). Where previously (in 1.6.5) this has caused the drone to land at its current position, it now send the drone to (0,0,0).

In short, I want this behavior: https://docs.px4.io/en/flight_modes/land.html

Can someone tell me the easiest way to check that the correct mode within PX4 is being triggered? I looked at the vehicle status from the ulog using pyulog, but I'm not sure which field, and in which set of data points, represents the mode or submode.

**To Reproduce**
Steps to reproduce the behavior:
1. Arm drone
2. Switch drone mode to OFFBOARD 
3. Send to a hover position offset in XY from the takeoff point
4. Switch drone to mode AUTO.LAND
5. Observe that drone returns to takeoff origin

Some other context:
* I do not start the navigator module, as I only use the commander for position control. From the code, it looks like the commander module handles the PX4_CUSTOM_SUB_MODE_AUTO_LAND, and I didn't think the navigator would handle landing unless it is RTL.
* Other blacklisted (through MAVROS config) modules: distance_sensor, ftp, hil, home_position, image_pub, obstacle_distance, rangefinder, safety_area, vfr_hud, vibration, waypoint

**Expected behavior**
Drone should land directly beneath its current position. PX4 should not set coordinates at which to land.

**Log Files and Screenshots**
https://logs.px4.io/plot_app?log=9b6cd93a-c36d-4609-9c74-64899d29b752

As you can see, PX4 enters LAND mode, but then sets a waypoint at 0,0,0.

**Drone (please complete the following information):**
Software and hardware versions as described [here]( https://docs.px4.io/en/flight_controller/snapdragon_flight_software_installation.html).

**Additional context**
* Before 1.8.0, I was on version 1.6.5. Autoland worked as expected in that version. However, I've made other configuration changes as well, so I can't point directly to a change in between those versions.

LorenzMeier, 2018-11-27:
@MaEtUgR @RomanBapst This sounds like a potential flight task regression of some sort.

MaEtUgR, 2018-11-28:
> Can someone tell me the easiest way to check that the correct mode within PX4 is being triggered?

Messages, fields you want to look at are:
- https://github.com/PX4/Firmware/blob/master/msg/commander_state.msg#L20
- https://github.com/PX4/Firmware/blob/master/msg/vehicle_status.msg#L54
Meanings of the values are in the uppercase dfines of the same message definition.

> 2. Switch drone mode to OFFBOARD
> 3. Send to a hover position offset in XY from the takeoff point
> 4. Switch drone to mode AUTO.LAND

This indeed sounds like an initialization problem of the executing task for landing. I've tested manual displacement followed by a land command a lot of times but never use offboard. Let me quickly have a look if it's easy to spot.

MaEtUgR, 2018-11-28:
I don't quickly see how this should be broken for offboard only. It could depend on MPC_AUTO_MODE being set to smooth or not maybe but shouldn't. Needs SITL debugging.

Note: @mrivi told me it works in her latest tests on master

mrivi, 2018-11-28:
@MaEtUgR @Stifael log from simulation https://logs.px4.io/plot_app?log=1c0c88b8-bc87-4c4c-8e4b-4c76a4943457

lading seems fine but I have navigator running

mrivi, 2018-11-28:
I tried killing navigator https://logs.px4.io/plot_app?log=fe66c076-924c-4c58-8787-5021790c4ef6

starts lading, then it starts rising again and then starts lading again.

Seanmatthews, 2018-11-28:
Thanks for all the replies. Let me unpack:

> @Seanmatthews What version are you using 1.8 stable? master branch?
I'm operating from the commit referenced by the 1.8.0 tag.

> I didn't see this statement. It's unsupported to do a proper land without navigator. Still the failsafe descend should land correctly on the spot.
To be clear, if I have navigator running it should land on the spot? Or do you mean that only the failsafe will land on the spot? Either way, do you all think it would be better to not transition to the AUTO.LAND submode if no navigator is running? To me, the current behavior seems unexpected and potentially disastrous.  

> Note: @mrivi told me it works in her latest tests on master
I'll give this a try and report back, but I won't be able to get to it until next week, unfortunately.

MaEtUgR, 2018-11-29:
@Seanmatthews 
> I'm operating from the commit referenced by the 1.8.0 tag.
That helps because we know what to test: the flight task architecture was not on there and it can't be a "flight task regression" 👍 

> better to not transition to the AUTO.LAND submode if no navigator is running?
AUTO.LAND is normally executed by the navigator with the current (and 1.8) flight stack state. Hence it is something unexpected to process a land without navigator and should lead to a failsafe reaction.

> if I have navigator running it should land on the spot?
It should theoretically land on the spot either way but that's more by accident because if you run the navigator it's the expected way and it should land on the spot. If you don't run the navigator it should trigger failsafe and instead of executing the non-valid output of the navigator land on the spot (or stay where it is).

> won't be able to get to it until next week
No worries, I'd encourage you to use the navigator if there's no hard constraint. We can still look at the failsafes when the navigator is not running.

MaEtUgR, 2019-03-02:
I'm closing this since any auto modes need to have navigator running. I can think of takeoff and land support without navigator in the future but there is no plan for this so far.

closed 2019-03-02 by MaEtUgR