PX4/PX4-Autopilot #10935: Testing procedures

sanderux, 2018-11-28:
As discussed on dev-call 28 nov 2018.
We should define procedures for test flights performed by the @PX4/testflights team.
This procedure should include
- How to communicate about the flight plan (flight cards?)
- What to check pre-flight
- What to watch during flight
- How to react to undesired or unexpected situations
- How to debrief

As a kickoff i propose:

1. always test manual flight is good in a safe environment
2. file a flight plan with the developer
3. if anything doesnt go as it should go back to manual and LAND asap
4. failures always to be reported before trying again, even if suspected hardware failure

@dagar @RomanBapst @LorenzMeier @mrpollo

dagar, 2018-11-28:
Existing flight cards - http://dev.px4.io/en/test_and_ci/test_flights.html

hamishwillee, 2018-11-29:
> We should add an exhaustive preflight checklist for each vehicle type. We want to catch as many problems as possible on the ground.
>
> General emergency procedures should also be documented (and practiced).

That sounds generic to any flyer. If so, then should be in the general guidance/best practise docs and linked to from test flights (rather than being part of test flights docs).

sanderux, 2018-11-30:
As reference here is the DeltaQuad pre-flight checklist:
https://docs.deltaquad.com/flight/pre-flight-checks.html

Tony3dr, 2018-11-30:
@sanderux for the pre-flight checklist we can use the ones you created. Just rearranged the order [https://docs.google.com/document/d/1jPdZ5hnGTXifoMysrEYlxnO5sp_9n2mzzOlZxlSll8o/edit#](url).

**What to watch during flight:**

- Take off

- Transition

-  Way points tracking

- Back Transition 

- RTL

- Land detection 

**How to debrief:**
Submit the[ issue](https://github.com/PX4/Firmware) and follow the below guidelines

- Vehicle: 

- Flight controller: 

- Firmware:

- Previous successful flight (date) 

Flight Plan (Photo example)**

Sequence of events:

Thoughts

hamishwillee, 2018-12-02:
@Tony3dr You plan on merging the above info into the docs? (i.e. I assume you will contact me if you need support).

Note to me, we probably should see if anything of this needs to be captured in https://docs.px4.io/en/flying/first_flight_guidelines.html

Tony3dr, 2018-12-03:
@hamishwilleeYes we will let you know once we have all the data ready. Thanks

RicardoM17, 2018-12-04:
@Tony3dr Please keep me in the loop as well. I've been keeping eyes on this post.

hamishwillee, 2025-03-27:
Removing docs label. No indication this requires any action.