PX4/PX4-Autopilot #10936: Gazebo Simulation - Rover in Offboard Mode

JoshMarino, 2018-11-28:
I am attempting to simulate the rover in Gazebo using ROS and Mavros, following [this example](https://dev.px4.io/en/ros/mavros_offboard.html). I am able to arm the rover, set a home position using its current GPS position, and switch to offboard mode. However, when I next try to send a position command {x: 10, y: 0, z: 0} to /mavros/setpoint_position/local at 20 Hz, the rover does not move. The only thing that happens is that the front wheels turn all the way to the left. Even tried changing the position command to {x:0 , y: 10, z: 0} and nothing happened either. 

I am able to control the iris model using the same approach in Gazebo using ROS and Mavros in Offboard mode, but the rover does not work. The only difference between the iris and rover simulations I am running is not calling /mavros/cmd/takeoff for the rover (although I tried it and it said unable to takeoff, try disarming and calling it again).

Any help would be appreciated.

SalimTerryLi, 2019-05-27:
It is because current rover controller doesn 't support navigating with local NED position. And there are still some bugs affecting navigating with attitude setpoint.

SalimTerryLi, 2019-05-27:
@julianoes Should we re-open this issue? Offboard mode is essential for an automatic rover. And it is not too difficult to fix.

julianoes, 2019-08-05:
@JoshMarino or @SalimTerryLi did you make any progress on this? It seems like you know what's missing and could potentially fix it. :smile:

PR #12765 (2019-08-21): Enable Offboard support for Rover position control
**Describe problem solved by the proposed pull request**
When streaming position setpoints with `SET_POSITION_TARGET_LOCAL_NED` to the rover, the rover is able to enter offboard mode but is not able to track the setpoint. This is because the rover position controller did not support offboard mode.

This Fixes #10936

**Test data / coverage**
The following log shows the log of testing in SITL by sending setpoints of [200m, 200m] in local coordinates.
https://review.px4.io/plot_app?log=4860d7a3-1366-4614-aee1-72f8dff95037

**Describe your prefer [... cut]
closed 2019-08-21 by dagar