PX4/PX4-Autopilot #10939: review RC parsing and NuttX parity bit

dagar, 2018-11-28:
SBUS uses an even parity bit. Is it actually being used with current NuttX?

 * https://github.com/uzh-rpg/rpg_quadrotor_control/wiki/SBUS-Protocol
 * https://github.com/PX4/Firmware/blob/master/src/lib/rc/sbus.cpp#L214-L218

dagar, 2019-02-27:
@davids5 what was the conclusion here?

davids5, 2019-02-27:
It is configured in the correct mode IIRC the error is not propagated nor that chatrater dropped. The issues that shed light on this was a mis configuration.  There is some "Framing" in the decoder  that may detect a problem or not. Are there any open issues that indicate SBUS related problems?

junwoo091400, 2022-07-27:
Is this still an issue?

davids5, 2022-07-27:
There is a PR in to test a fix https://github.com/PX4/PX4-Autopilot/pull/19446 but I am not sure if the testing that has been done is testing the right things or reporting the right problems.