PX4/PX4-Autopilot #10951: Mag sensor inconsistent error

DavidMackie, 2018-11-29:
Using the latest stable release of PX4 and latest stable release of QGroundControl, "Mag sensor inconsistent" errors pop-up as pre-flight checks fail.
When doing the calibration with a 2018 August QGC build, the problem disappears.

## Expected Behavior
After a successful Firmware update (PX4 stable), and a successful sensor calibration (for a Deltaquad airframe) the magnetometer inconsistency check has so far been working properly.

## Current Behavior
When using the latest QGC stable build, the inconsistency pops up and pre-flight checks fail, causing the arming to fail.

## Steps to Reproduce:

The orientation of the board must be 90 degrees roll, 90 degrees yaw.

1. With the latest stable version of QGC for Ubuntu flash the latest stable firmware.
1. Calibrate everything from fresh with the latest version of QGC for Android (Or Ubuntu)
1. Restart the drone
1. Mag sensor inconsistent and 90 degree heading offset is seen

## System Information

* Ubuntu 18.04
* QGC build which drops error: QGC Development HEAD:1bf12b5b3 2018-11-13
* QGC build which fixes error: QGC Development HEAD:9cb4f2ddc 2017-08-07
* Flight Controller: Dropix v2.1 from Drotek
* Primary Compass: Ublox NEO-M8N GPS + LIS3MDL compass
* Autopilot (with version): PX4 1.8.1 stable

## Additional related problem
Another issue that pops up is when the measured compass heading is about 90 off true heading. Calibration was done in relatively empty fields. The offset seemed to be consistent
This is either PX4 estimator related or calibration related.

## Log Files and Screenshots
Short test where the drone is pointing close to North direction:
https://review.px4.io/plot_app?log=17da1d39-138e-4495-b304-4630c6c58a76

A screenshot of the error when it pops up on on Android tablet:
[image]

Thanks for any help!

AlexisTM, 2018-12-06:
Hey;

Few more steps you can use to debug it yourself:
- Enable SDLOG_PROFILE enable "sensor comparison". 
- Open the log with FlightPlot (java application)
- In that log, you'll have the raw sensor data, plot the data for all XYZ axis and you'll see if they are consistent or if there is a problem (such as wrong orientation).
- Double check SENS_BOARD_ROT parameter (currently 18)

YCH188, 2019-01-05:
@DavidMackie Have you solve the problem, I also have "Mag sensor inconsistent" errors and the controler cannot be started because of the problem.

Syavick, 2019-01-09:
> @DavidMackie Have you solve the problem, I also have "Mag sensor inconsistent" errors and the controler cannot be started because of the problem.

I've disabled cal_mag1_en and it worked for me.

closed 2019-01-10 by DavidMackie