PX4/PX4-Autopilot #10953: Compass Sensor 0 missing PixHawk 2

yvescharles, 2018-11-30:
Compass Sensor 0 still missing after multiples successfull build run and firmware update in QGC bases on new airframe geometry toml + doc and mixers only 'see my pull request for complete files)

- Seem to be not hardware issue : Problem is the same with any kind of airfraime but desappear when revert to stable firmware (probably driver ?)

Please see here to folow the steps  http://discuss.px4.io/t/thrust-and-torque-constants-in-toml-geometry-description-and-sitl-gazebo-models/5986/13?u=yvescharles

yvescharles, 2018-11-30:
find only one matching message in source code here : https://github.com/PX4/Firmware/blob/master/src/modules/commander/PreflightCheck.cpp#L132

KeyneZzz, 2018-12-03:
Meet the same problem, and not only compass, the sensors on board all seems unavailable: compass, accel, gyro.
In my view, there might be some problem with the newly organized makefiles

yvescharles, 2018-12-03:
@mhkabir when/what do you mean on master since I was running sveral build on my fork/master which is even the same from origin ? are you using à specific git tag on px4/fimware source code ?
Thanks for your support

thanks also @KeyneZzz

mhkabir, 2018-12-03:
Pull master as of today and it should be fixed. This is the hash : https://github.com/PX4/Firmware/commit/c6a502f150c9f3865a453f131cbc6fb35c82384c

nesrine1211, 2019-05-23:
> Thanks @mhkabir this work well !

I'm facing the same problem but didn't manage to solve it. 
I have done: 
git checkout c6a502f150
git fetch origin
git merge origin/master. 
What am I doing wrong?
Thank you