PX4/PX4-Autopilot #10961: Master branch firmware (fmu v3) cannot drive all sensors

cuhome, 2018-12-01:
I downloaded the master firmware from qgc and found that it is not normal. It only recognizes ms5611 and other sensors are not being driven normally.
[image]
 I found a strange problem with a version number of 1.80dev instead of 1.81 or 1.82. But it shouldn't be an old firmware because it has added a new rgb driver.
fmu_v3
I tested pixhawk, cube, and pixhack, and have the same problem*

jinger26, 2018-12-01:
@dagar @davids5

dagar, 2018-12-01:
It's the `HW arch` string.

After the board refactor the fmu-v3 string came from the board model, instead of the previously hardcoded FMU_V2. As a result it's not entering rc.sensors.

dagar, 2018-12-01:
This is fixed in https://github.com/PX4/Firmware/pull/10960, which gets rid of the majority of the startup version checks and in general should make things simpler.

dagar, 2018-12-01:
Fixed by https://github.com/PX4/Firmware/pull/10960

closed 2018-12-01 by dagar

cjk7713, 2018-12-04:
#10960