PX4/PX4-Autopilot #10965: libm linking issue (undefined reference to `__errno')

christianrauch, 2018-12-02:
**Describe the bug**
I am porting to a STM32 F1 (STM32F103CBT6) and got linking issues with `asin` and `sqrt`:
```
[...]/gcc-arm-none-eabi-7-2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/libm.a(lib_a-wf_asin.o): In function `asinf':
wf_asin.c:(.text.asinf+0x7c): undefined reference to `__errno'
wf_asin.c:(.text.asinf+0x86): undefined reference to `__errno'
[...]/gcc-arm-none-eabi-7-2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/libm.a(lib_a-wf_sqrt.o): In function `sqrtf':
wf_sqrt.c:(.text.sqrtf+0x60): undefined reference to `__errno'
wf_sqrt.c:(.text.sqrtf+0x8c): undefined reference to `__errno'
[...]/gcc-arm-none-eabi-7-2017-q4-major/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/libm.a(lib_a-math_errf.o): In function `with_errnof':
math_errf.c:(.text.with_errnof+0x6): undefined reference to `__errno'
```
I can resolve this issue by linking `libm` (set `CONFIG_LIBM`).
This in turn, however, introduces issues with the definition of `isfinite`:
https://github.com/PX4/Firmware/blob/161cf7f5d92abba0b1b862d7aaa0939cb69d42bb/src/platforms/px4_defines.h#L137-L142
since it is using defines and functions from header `arch/math.h`, which are only available when `CONFIG_ARCH_MATH_H` is set. In fact, definitions from `arch/math.h` are used at some other places too, without checking that `CONFIG_ARCH_MATH_H` is actually set.

I can set `CONFIG_LIBM` and include `arch/math.h` manually in `px4_defines.h`. But this doesn't seem to be right thing to do, since both are not supposed to be used at the same time.

**To Reproduce**
Unzip [naze32.zip](https://github.com/PX4/Firmware/files/2636418/naze32.zip) to the boards folder and compile via: `make naze32_rev6_full_default`.

Is `CONFIG_ARCH_MATH_H` assumed to be always set (instead of `CONFIG_LIBM`)? If yes, how do I resolve the linking issues? If no, how do I use `px4_defines.h` without the defines from `arch/math.h`?

christianrauch, 2018-12-02:
That's good to know. I added `-fno-math-errno` to `FLAGS` and they now appear in the verbose log. But I am still getting the `__errno` linking errors.

davids5, 2018-12-03:
There is more to this. PX4 is built with it's on math header. (There is a [REJECTED] commit in the PX4 nuttx)

Have a look at the porting guide and the http://nuttx.org/doku.php?id=documentation:configvars
to gain an understanding of what is a real issue as apposed to mis configuration.  

```
1.14.2 CONFIG_LIBM: Standard Math library
Type: Boolean
Default: n
Dependencies: !CONFIG_ARCH_MATH_H
Kconfig file: ./libs/libc/math/Kconfig
By default, no math library will be provided by NuttX. In this this case, it is assumed that (1) no math library is required, or (2) you will be using the math.h header file and the libm library provided by your toolchain.

This is may be a very good choice is possible because your toolchain may have have a highly optimized version of libm.

Another possibility is that you have a custom, architecture-specific math libary and that the corresponding math.h file resides at arch/<architecture>/include/math.h. The option is selected via ARCH_MATH_H. If ARCH_MATH_H is selected,then the include/nuttx/libmath.h header file will be copied to include/math.h where it can be used by your applications.

If ARCH_MATH_H is not defined, then this option can be selected to build a generic, math library built into NuttX. This math library comes from the Rhombus OS and was written by Nick Johnson. The Rhombus OS math library port was contributed by Darcy Gong.
```

christianrauch, 2018-12-04:
By grepping over the linker scripts (`ld.script`) I found a hack that was used in all other linker scripts to define `__errno`:

		/*
		 * This is a hack to make the newlib libm __errno() call
		 * use the NuttX get_errno_ptr() function.
		 */
		__errno = get_errno_ptr;

This hack is not present in the upstream linker scripts from NuttX, that I based my port on.

Adding this to the linker script solves the linker issue. The solution is similar to what was suggested in https://nuttx.yahoogroups.narkive.com/t3Eii1bA/integrating-math-h-and-libc-a (`# define errno *get_errno_ptr()`) .

closed 2018-12-04 by christianrauch