PX4/PX4-Autopilot #10969: Unable to modify configuration settings for Rover - QGC says vehicle is in flight.

rsmartmachine, 2018-12-03:
When using pixhawk 4 with latest rover firmware QGC does not let the user modify any settings. Says (Disabled while the vehicle is flying )

DonLakeFlyer, 2018-12-03:
You can see the related QGC issues here: https://github.com/mavlink/qgroundcontrol/issues/7021/

The problem is that Rover is always reporting MAV_LANDED_STATE_IN_AIR. QGC uses this flag in various parts of the UI to indicate that the vehicle is actively doing something. Not just sitting on the ground doing nothing. Various vehicle setup pages and also guided actions are locked out when in this state. 

Some examples of non-vehicle setup cases where this state is used:
* Disarming changes to emergency stop.
* RTL and Pause buttons will be enabled
* ...

So for Rover would be good to maintain the same concepts so QGC can still run the same UI logic.

dlarue, 2018-12-17:
Running into the same issue trying to setup an OmnibusF4 Pro for use on a DIYRoboCar autonomous vehicle.  And the problem is, once you get this you can't change the airframe back to a flying vehicle so you can calibrate or otherwise set up the configuration.

DonLakeFlyer, 2018-12-17:
I'll tweak QGC to special case rover for now to allow setup screens. But that won't fix the problem since other part of QGC will fail without the status reporting.

dlarue, 2018-12-20:
I've tried the nightly builds the past 2 days and it still thinks the rover is flying and i still can't calibrate sensors or do much of anything in QGC.

dlarue, 2018-12-23:
in settings->general I enabled saving telemetry logs even when not armed. Here's the log:
[2018-12-22 19-00-37.tlog.zip](https://github.com/PX4/Firmware/files/2705719/2018-12-22.19-00-37.tlog.zip)

dlarue, 2019-03-23:
It shouldn't be stale... Rover capability is a growing market/ segment and even moreso with CNN/AI taking over.

Jaeyoung-Lim, 2019-03-23:
@dlarue Would be really cool if the rover airframe can be used in DIYRobocar community!

dlarue, 2019-03-23:
Yes @Jaeyoung-Lim , that was/is my intention. I will get back to it soon. I already have ROS Melodic and a donkey fork running side-by-side on an rPi3B so it's just a matter of rewiring and publishing/subscribing to steering and throttle topics.

dlarue, 2019-08-16:
so this didn't get fixed?   Well, rovers are probably just a fad anyways...

DonLakeFlyer, 2019-08-16:
@dlarue QGC has a workaround fix in it which makes this work. That said, this should still be fixed on the firmware side.

julianoes, 2019-08-19:
I think this has been fixed but I'm re-opening this and assigning @ItsTimmy to check. I'm pretty sure he has done that recently.

dlarue, 2019-11-17:
still waiting for status that it was at some point fixed. Thanks.

ItsTimmy, 2019-11-29:
@dlarue As far as I can tell, this is currently fixed. The rover says it is "in flight" as long as it is armed. As soon as it disarms, it reports being "landed". If you have any issues, try disarming.

closed 2019-11-29 by ItsTimmy