PX4/PX4-Autopilot #10975: Poor flight path following during FW final auto landing approach

zanemountcastle, 2018-12-03:
**Describe the bug**
Although the FW plane flies auto missions exceptionally well, the plane struggles to follow a straight path on the final landing approach--and only on the final landing approach. Instead, it swerves (sometimes very aggressively) as it descends.

I have ruled out the plane losing control due to stalling as the stall speed is about 11 m/s. The plane remains at about 13 m/s during the landing approach. See the logs below.

It seems as though the plane is overcorrecting its yaw angle. However, this seems to happen **only** during the landing approach. While cruising, the plane does very well at compensating for wind and remains very stable.

**To Reproduce**
Steps to reproduce the behavior:
The only step to reproduce the behavior is to fly an auto mission including an auto landing.

**Expected behavior**
The plane should fly in a straight line throughout its final approach and not swerve.

**Log Files and Screenshots**
Here are three separate flights with the same behavior. These flights were all conducted at the same site in roughly the same conditions, but this behavior has been experienced in all wind conditions, terrain, etc. that I have tested it in.

https://logs.px4.io/plot_app?log=94994303-2e07-4a8d-8796-04ffb223ea72
https://logs.px4.io/plot_app?log=ac1454a6-9d4f-4c8c-8cde-a53ccad1a2a9
https://logs.px4.io/plot_app?log=aacc386a-5f6d-4945-bda2-143f521e23fd

Add screenshots to help explain your problem.

[image]
[image]

**Drone (please complete the following information):**
I am using a Finwing Penguin FPV airframe with a Pixhawk 2.1 running PX4 1.8.1.

**Additional context**
The plane has flaps enabled which do deploy as the plane is coming in for landing. However, these flaps don't seem to affect the flight path during any other portion of the auto missions so I doubt that they are the cause.

Antiheavy, 2018-12-03:
Hello @zanemountcastle .  I see a couple issues based on a quick glance at your logs:

1) you are using the default inner loop gains and they seem to be driving your yaw and roll way too hard.  Look at your yaw and roll angles and yaw and roll rates in the Flight Review logs, the estimated values are oscillating strongly above and below the setpoints.  You could get a lot better flight control if you tuned your gains.  Gains tuning is often more art than science on opensource autopilot systems - be sure you have a safety pilot with manual control ready!
https://docs.px4.io/en/config_fw/pid_tuning_guide_fixedwing.html

2) You have Flaperon's enabled which I think are causing the flaps to also be acting as additional ailerons on landing which is worsening your roll control, it seems to be lagging badly behind the setpoint.
[image]
[image]

My recommendation:
Step 1 is gains tuning.
Step 2 is Flaps and Flaperon tuning.

(P.S. you might consider RTL_TYPE = 1 if you want the vehicle to immediately break off mission and come straight into the landing pattern, up to you).

zanemountcastle, 2018-12-04:
Thanks @Antiheavy!

Sorry that I said flaps, I meant flaperons--no flaps, just flaperons. That said, during landing, the plane isn't rolling much. Instead, it's the yaw overcorrection that's the issue.

I'll try tuning the gains and update here ASAP.

closed 2019-03-18 by stale[bot]