PX4/PX4-Autopilot #10976: CUAV V5 debug: Board 105 is not supported

cuhome, 2018-12-04:
Firmware: master branch firmware (fmu v5) downloaded online on qgc
problem:Board 105 is not supported（The fault has been fixed in the 1.80/1.81stable firmware, but the new master firmware reproduces this problem.）
[image]

davids5, 2018-12-04:
@jinger26 

This is not a bug or fault or regression - Version 1.8.0, 1.81 did not check the hardware revision nor provide a manifest. 

Product was shipped without getting a REV and VER assigned by PX4. 

As shipped it is not supported and I believe that is because SBUS RC does not work because it is wired differently than the FMUv5 pin out. 

Please discuss with @LorenzMeier and let me know how you would like to resolve this.

davids5, 2018-12-04:
@cuhome - please test with https://github.com/PX4/Firmware/pull/10982

closed 2019-03-04 by davids5