PX4/PX4-Autopilot #10979: MC_YAW_FF parameter missing from firmware

glegarda, 2018-12-04:
Hello,

I followed the steps described in the Developer Guide to install a development environment on Ubuntu LTS and build the latest version of px4_fmu-v3_default. I uploaded the firmware to my HKPilot32 and downloaded the latest QGC daily build. However, when I go on the PID tuning tab in the parameters, I get the following error:

"Parameters are missing from firmware. You may be running a version of firmware QGC does not work correctly with or your firmware has a bug in it. Missing params: -1: MC_YAW_FF"

The parameter is indeed missing.

I would stick to the stable release, but QGC does not recognise my X4R-SB receiver automatically, so I need to install the firmware manually with the driver activated.

Any help fixing this would be very much appreciated.

hamishwillee, 2018-12-04:
Param was deleted from Firmware in May: https://github.com/PX4/Firmware/commit/cc0bc051560b432d0f359ce96ca38b3f46748fc9

@DonLakeFlyer FYI

glegarda, 2018-12-04:
I completely missed that. Thanks @hamishwillee.

closed 2018-12-04 by mhkabir