PX4/PX4-Autopilot #10980: Issue with Master

Tony3dr, 2018-12-04:
**Describe the bug**
 Issue with Master, the ESC's don't initialize. Tested a couple vehicles and the i/o (specifically the safety switch) is non-responsive. Flashed master from qgc e69049e onto a pixhawk mini and the safety switch seems to be working just fine. No, go on all other platforms, including the pixhawk mini being used
the safety switch on the pixhawk mini (v3), flashes faster than normal and is non-responsive
same for the pixhawk 4 (v5) which has a safety switch in the gps module.
The pixracer (v4) that has an external safety switch shows no activity (no flashing). 
Some parameters seem to be creating a conflict, resetting the airframe seems to fix the board we tested but loading old parameters puts it in a non-responsive state. We'll recalibrate all vehicles and should be good from there

Noticed in master, there is no longer a circuit breaker for the safety switch (IO_SAFETY_C...), could this be relevant?

Last working commit- ef178730
Tested platforms- Pixhawk mini, Pixhawk 4, Pixhawk 4 mini, Pixracer, Pihawk Pro
[image]

@dagar @mrpollo @santiago3dr

dagar, 2018-12-04:
Can you link to the exact binary?

Tony3dr, 2018-12-04:
current binary:
https://github.com/PX4/Firmware/commit/e69049ec4183d820c6f716be1577df2ad07194b8

last working binary
https://github.com/PX4/Firmware/commit/ef178730b2f2cb8acc635152bf3c9f3057d24ed2

This has been somewhat resolved, **resetting the airframe seems to fix the board**

Tony3dr, 2018-12-04:
We reset the parameters & Installed stable, seems to be working. Same steps, reset parameters, installed Master, not working. That means that todays master is changing some parameter that causes the failure @dagar

santiago3dr, 2018-12-04:
edit:
flashed master e69049e, board non-responsive
reset airframe, board responsive
loaded parameters from previous flights: board non-responsive

dagar, 2018-12-04:
Fixed in https://github.com/PX4/Firmware/pull/10983

It was only a problem for certain multicopter airframes that referenced 4001 (Generic Quadrotor x).

commit ecbf6ea7 (2018-12-05): ROMFS multicopter airframes don't call other airframes (4001/4002)

 - fixes #10980
closed 2018-12-05 by dagar