PX4/PX4-Autopilot #10981: fmu-v5: ERROR [pwm] can't open /dev/pwm_output0

JMshare, 2018-12-04:
**Describe the bug**
All the ESCs start beeping on startup and never stop if the battery is connected.
In the QGroundControl Mavlink Console I type pwm info and get:
`ERROR [pwm] can't open /dev/pwm_output0`

My board is px4_fmu-v5. Compiling and uploading from the current master [https://github.com/PX4/Firmware/commit/e69049ec4183d820c6f716be1577df2ad07194b8](url)

I can say that it worked well around two weeks ago, and it works without this issue on the v1.8.2 release. 
[image]

**Drone (please complete the following information):**
- Pixhawk 4 on DJI F450 quad_x airframe

dagar, 2018-12-04:
@JMshare this should be fixed in https://github.com/PX4/Firmware/pull/10983
Thanks for reporting.

dagar, 2018-12-05:
Fixed in https://github.com/PX4/Firmware/pull/10983

closed 2018-12-05 by dagar