PX4/PX4-Autopilot #10984: UAVCAN SETUP

Avysuarez, 2018-12-04:
@dagar @mhkabir @RomanBapst @mrpollo @Tony3dr 
I have some issues with the uavcan setup, I followed the instructions for the setup and I've only managed to fly like this. 
These are the parameters of the setup. Any suggestions? 
UAVCAN Test
Assigment ESC's
Motor 1: 0
Motor 2: 1
Motor 3: 2
Motor 4: 3
Nodeid
Motor 1: 11
Motor 2: 12
Motor 3: 13
Motor 4: 14

[image]
[image]
[image]
This is a flight video.
https://youtu.be/uKX-MOZG7ww

Any suggestions?

jinger26, 2018-12-05:
@pavel-kirienko ?

mhkabir, 2018-12-05:
The vehicle PIDs are tuned too high compared to the response characteristics of the ESC.

Seems like the either Myxas are much slower than the standard PWM ESCs the system is tuned with, or there is a massive latency somewhere in the chain.

DanielePettenuzzo, 2018-12-05:
We had the same behaviour when we started. On our setup we had to increase the current_ctl_bw parameter to 0.12 after @pavel-kirienko 's suggestion. This will decrease the slew rate. If you still have this behaviour then you need to slightly detune the vehicle so that we don't demand as much from the esc's. 

Motor indexes and node ids look good.

Moreover can you please let Pavel know about any information that you think is missing in the myxa documentation.

pavel-kirienko, 2018-12-06:
Increase `m.voltage_ramp` (start with 30) and set `m.current_ctl_bw` to 0.1. More info in the docs.

Tony3dr, 2018-12-11:
We tested the UAVCAN copter but this time on the field, it flew better but it does not work in position and mission, the GPS registers up 24 sats. Even then it does not work.

This is a log of the flight:
https://review.px4.io/plot_app?log=5b3f576a-b07d-48ae-baab-d58f2561b710

Video of the flight:
https://www.youtube.com/watch?v=IUT5dC-MriM&feature=youtu.be

@dagar @mhkabir @pavel-kirienko @RomanBapst can someone check out the log and provide feedback.

Thanks, everyone.

mhkabir, 2018-12-12:
Vibrations are pretty bad

closed 2019-01-23 by Tony3dr