PX4/PX4-Autopilot #10988: HITL jMAVSim: MSG: Preflight Fail: Position unknown

JMshare, 2018-12-05:
**Describe the bug**
The HITL with jMAVSim keeps printing 
`MSG: Preflight Fail: Position unknown.`.

It seems it lets me arm but immediately it says 
`MSG: Takeoff detected`,
`MSG: CRITICAL NAVIGATION FAILURE - CHECK SENSOR CALIBR`.

The sensors readings in the GCS are all over the place.
[image]

Running on fmu-v5 with the current master [https://github.com/PX4/Firmware/commit/a5fa3eda302ca0ecd8b6938b3fbe940ae659a76c](url), following the instructions on [https://dev.px4.io/en/simulation/hitl.html](url), calling `./Tools/jmavsim_run.sh -q -d /dev/tty.usbmodem1 -b 921600 -r 250` on MAC.

I am using the HIL Quadcopter X airframe and QGroundControl v3.4.4.

I think it worked fine few days ago but not sure if I used the same QGC version then.

E666GT, 2019-01-13:
I meet the same error as you did. Have you get Any solution to it? Thanks

JMshare, 2019-01-13:
Unfortunately not

E666GT, 2019-01-13:
Thank you, if I got any progress I'll let you know

ironman03, 2019-06-21:
I have same the proplem? you can help me? please

julianoes, 2019-06-23:
When it happens, does the frame rate shown in the bottom left go to around 5 fps?

ironman03, 2019-06-24:
I solved the problem, because I was not calibratedcompass properly. 
After calib, the copter has arm and no error.

julianoes, 2019-07-01:
Great thanks for reporting this.

closed 2019-07-01 by julianoes

JMshare, 2019-07-31:
> I solved the problem, because I was not calibrated compass properly.
> After calib, the copter has arm and no error.

You are saying I need to calibrate compass for HITL?