PX4/PX4-Autopilot #10991: [Critical Crash] Failsafe overwritten with Killswitch on RC Lost

TobiMiller, 2018-12-05:
The following behavior in the firmware is dangerous and not understandable from user-side. Why should i want that the kill switch is engaged on RC Lost? makes no sense.
We flew some surveys in range of our remote without any problem. On the last survey the remote lost the connection to the drone. The drone was set to RTL as safety response on RC lost. The problem: We also assign the kill switch. Sidenote: In dangerous situation you pull a switch on the side of the remote instead of pushing. So i decided to invert the channel on remote to get displayed the correct response in QGroundControl. When you get out of range the channel flipped on firmware-side back to full. So the Octocopter dropped from sky like a stone including our sensorbox (Velodyne, radar, camera, zed camera) instead of coming back. 
RESULT: We are lucky that "just" the DJI S1000 is completly broken and needs to be replaced.

**To Reproduce**
Set Safety on RC Lost to RTL.
Set Kill switch to channel.
Invert Killswitch on your remote.
Fly out of range.

**Expected behavior**
UAV will come back. --> RTL --> should do what you set in safety rules

**Log Files**
https://review.px4.io/plot_app?log=e49ff554-c20a-4d47-a9ef-95091b8be331

dagar, 2018-12-05:
Sorry to hear about your crash. I see that RTL was never triggered, but it was because RC was never considered lost, not because of the kill switch.

[image]

Zooming in you can see the Pixhawk continued to receive signals from the RC receiver. When RC was lost the receiver continued to output.

[image]

What RC equipment are you using? What's it configured to do when signal is lost?

hamishwillee, 2018-12-05:
@dagar AFAIK docs don't say anything about the receiver having to be "specially" configured to report RC loss?
- What does PX4 "expects" when RC is lost? (and is this in any way tunable PX4 side?)
- Are most receivers tuned the right way "out of the box"? Can we list some examples of receivers that do work out of the box, or otherwise give a hint to users for where they look for appropriate receiver configuration (ie search terms they might look for in manual to find correct mode). 

Two sections we might update:
- https://docs.px4.io/en/getting_started/rc_transmitter_receiver.html
- https://docs.px4.io/en/config/safety.html#rc-loss-failsafe

TobiMiller, 2018-12-06:
@dagar 
Then tell me why kill switch was engaged ?! when we were at ground and the remote was switched off  qgroundcontrol was telling that kill switch was engaged again.

and i saw in the channel mapping in qgroundcontrol the channel switching to "1" when the remote was switched off.

why is there no channel inverting on qgroundcontrol as checkbox when it is not considered to be done on remote side?

mhkabir, 2018-12-06:
@TobiMiller Let's take a step back. From your log :

dagar, 2018-12-06:
@TobiMiller I believe that the root cause is your RC failsafe configuration. When RC is lost, or the transmitter is powered off the RC receiver sends these fixed values. You can see this in your log.

[image]

Then combined with the way you've inverted the switch when your RC is working it sends the switch position configured for "kill switch disengaged". When RC is lost the receiver puts the killswitch channel to the value corresponding to "kill switch engaged". 

To reverse the channel from the flight controllers perspective look at the `RCX_REV` parameters. However, that's not the core problem here. You need to look at your RC equipment and configure the failsafe mode to  stop sending data. In a Taranis this is `Failsafe Mode` = `Not Set` or `No Pulses`.

TobiMiller, 2018-12-10:
@dagar 
Thanks for your help understanding what is going on. My understanding of failsafe was wrong. I didnt know about the failsaife in the receiver.

dagar, 2018-12-10:
@TobiMiller not a problem, let's take it as an opportunity to improve the documentation.

closed 2019-03-24 by stale[bot]