PX4/PX4-Autopilot #10994: SD Card maximum capacity.

Louisdirk, 2018-12-06:
@LorenzMeier @simonegu @bkueng
I and trying to use 64GB Sandisk Extreme PRO U3 but failed to mount.
So what's the maximum capacity supported in NuttX?
Thank you.

hamishwillee, 2018-12-06:
I don't know, but FYI, these are recommended: http://dev.px4.io/en/log/logging.html#sd-cards

Louisdirk, 2018-12-06:
Yes, I saw that. I put 32GB one in the same board it's OK.
So I am confusing whether NuttX support higher capacity SDcard as my project requires.

bkueng, 2018-12-06:
@davids5 can probably help.

davids5, 2018-12-06:
I would say that nuttx is in compliance with SD Memory Card Specifications Version 2.0
So 32GB

`The Secure Digital eXtended Capacity (SDXC) format, announced in January 2009 and defined in version 3.01 of the SD specification, supports cards up to 2 TB (2048 GB), compared to a limit of 32 GB for SDHC cards in the SD 2.0 specification.`

hamishwillee, 2018-12-07:
FYI, added information about SD card max size to the [user guide basic concepts](https://github.com/PX4/px4_user_guide/pull/399) and Devguide [logging topic](https://github.com/PX4/Devguide/pull/686).

hamishwillee, 2018-12-07:
Closing as answered.

closed 2018-12-07 by hamishwillee

Louisdirk, 2018-12-07:
Thank you all.
@davids5 When will version 3.01 of the SD specification be supported? Or can you please give me some advice if I implement it by myself?

mhkabir, 2018-12-07:
@Louisdirk what is your use case?

Louisdirk, 2018-12-10:
@davids5 That sounds enormous work. Let me start to understand the details of the Specification.
@mhkabir Just a case needs a mount of logs.

Thank you.