PX4/PX4-Autopilot #10997: RC Can't generate PWM output and rotor not spin in PX4 pixhawk1

E666GT, 2018-12-06:
I use PX4 in pixhawk1.
Problem Description:
My RC controller can arm and disarm the PX4, but can't make rotor rotate.

Test Results: (may help you to locate where my problem is)
0. nsh> px4io start. I start px4io to use RC, otherwise RC can't be used.
1.I use nsh ' pwm test -p 1000 -a ' to make ESC know I'm starting. Then use ' pwm test -p 1200 -a ' or' pwm test -p 2000 -a ' to make rotor spin. It Spins! But, with RC controller, it makes no spin.
2. APM can successfully works on my pixhawk1.It means rotor, esc , pixhawk works. But it with PX4 does not work.
3. My pixhawk1 has 2 mb flash. Both px4_fmu-v2 and v3 were tried.
4. After arming, RC controller Can't make it spin. but give pwm test, it spins.Use RC no spins.
5. PX4io status shows change with RC controller changing. (in the line of 8 raw R/C inputs) But 
controls 0: 0 0 0 0 0 0 0 0
controls 1: 0 0 0 0 0 0 0 0
controls 2: 0 0 0 0 0 0 0 0
controls 3: 0 0 0 0 0 0 0 0
6. RC calibration finished. Makes no difference. 
7.Rotor can spins by using nsh> pwm test -p 1400 -a, so it means no error in ESC and rotor.

 I have no idea what the problem is.  It cost me days on this issue. I really need a support.

Best Appreciation for your help!

dagar, 2018-12-06:
Can you verify the exact version of PX4 you flashed? You can run `ver all` in the nsh console.

Which airframe did you configure (box in QGC)? This is the SYS_AUTOSTART parameter.

E666GT, 2018-12-06:
Hello @dagar  !
Thanks for your focus on my question.

Here's 'ver all' result:
`HW arch: PX4_FMU_V3
HW type: V2
HW version: 0x00090008
HW revision: 0x00000000
FW git-hash: 161cf7f5d92abba0b1b862d7aaa0939cb69d42bb
FW version: 1.8.0 0 (17301504)
FW git-branch: master
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: 166d898c70141d7ddb179b5ead1c5ea726574411
Build datetime: Dec  6 2018 12:42:05
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 0001000000003436333932355110001f0032
MCU: STM32F42x, rev. 3
`

I use AIrFrame "Quadrotor x"  "DJI Flame Wheel F450".

I'll appreciate to hear more instruction from you!
Thanks a lot !

E666GT, 2018-12-13:
I've solved this issue by update the Firmware with QGC.
I used to believe when I git pix4 I get the newest release, But maybe I'm wrong.
I use QGC- Firmware to get the newest verision 1.8.1
And it works

closed 2018-12-13 by E666GT