PX4/PX4-Autopilot #11003: FMUv2 Nuttx build regression

julianoes, 2018-12-07:
I'm sometimes getting a build error when doing a clean build.
I started looking into it but it's not obvious to me what happens.

FYI: @bkueng @dagar 

```
CCACHE_DISABLE=1 make px4_fmu-v2_default
-- PX4 config file: /home/julianoes/src/Firmware/boards/px4/fmu-v2/default.cmake
-- PX4 config: px4_fmu-v2_default
-- PX4 platform: nuttx
-- PX4 version: v1.8.0-1912-gdc49e259b3
-- cmake build type: MinSizeRel
-- The CXX compiler identification is GNU 7.2.1
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/julianoes/Software/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working CXX compiler: /home/julianoes/Software/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /home/julianoes/Software/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /home/julianoes/Software/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working C compiler: /home/julianoes/Software/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.15") 
-- Found PY_jinja2: /usr/lib/python2.7/site-packages/jinja2  
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-608-gacde4eb
-- Building and including px4_io-v2_default
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/julianoes/src/Firmware/build/px4_fmu-v2_default
Scanning dependencies of target git_nuttx_apps
Scanning dependencies of target px4io_firmware
Scanning dependencies of target git_nuttx
Scanning dependencies of target uorb_headers
[  0%] git submodule platforms/nuttx/NuttX/nuttx
[  0%] Creating directories for 'px4io_firmware'
[  0%] git submodule platforms/nuttx/NuttX/apps
[  0%] Generating uORB topic headers
[  0%] Built target git_nuttx
[  0%] Built target git_nuttx_apps
Scanning dependencies of target mixer_gen
Scanning dependencies of target mixer_gen_6dof
[  0%] Generating mixer_multirotor_6dof.generated.h
[  0%] Generating mixer_multirotor.generated.h
[  0%] No download step for 'px4io_firmware'
[  0%] No patch step for 'px4io_firmware'
[  1%] No update step for 'px4io_firmware'
[  1%] Generating mixer_multirotor_normalized.generated.h
[  1%] Performing configure step for 'px4io_firmware'
[  1%] Built target mixer_gen_6dof
Scanning dependencies of target ver_gen
-- PX4 config file: /home/julianoes/src/Firmware/boards/px4/io-v2/default.cmake
[  1%] Generating git version header
-- PX4 config: px4_io-v2_default
-- PX4 platform: nuttx
-- PX4 version: v1.8.0-1912-gdc49e259b3
-- cmake build type: MinSizeRel
[  1%] Built target mixer_gen
-- The CXX compiler identification is GNU 7.2.1
Scanning dependencies of target git_gps_devices
[  1%] git submodule src/drivers/gps/devices
-- The C compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /home/julianoes/Software/gcc-arm-none-eabi-7-2017-q4-major/bin/arm-none-eabi-gcc
[... 253 lines cut ...]
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/net/route': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/net/sixlowpan': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/net/tcp': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/net/udp': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/net/usrsock': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/net/utils': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/environ': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/errno': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/group': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/init': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/irq': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/module': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/mqueue': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/semaphore': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/signal': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/task': File exists
make[3]: *** [NuttX/CMakeFiles/nuttx_configs_build.dir/build.make:12610: NuttX/nuttx_copy.stamp] Error 1
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/timer': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/wdog': File exists
cp: cannot create directory 'build/px4_fmu-v2_default/NuttX/nuttx/sched/wqueue': File exists
make[3]: *** [NuttX/CMakeFiles/nuttx_apps_build.dir/build.make:10256: NuttX/nuttx_copy.stamp] Error 1
make[2]: *** [CMakeFiles/Makefile2:8094: NuttX/CMakeFiles/nuttx_configs_build.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/Makefile2:8028: NuttX/CMakeFiles/nuttx_apps_build.dir/all] Error 2
```

dagar, 2018-12-07:
Weird, I'll take a look. I suppose we should get a couple builds on Jenkins that don't use Ninja.

julianoes, 2018-12-07:
That would make sense. (Ceterum censeo ninja shouldn't be used.)

julianoes, 2018-12-07:
This is another warning/error that I get sometimes:
```
[  4%] Copying NuttX config nsh and configuring
[  4%] Generating nuttx/include/nuttx/version.h, nuttx/include/nuttx/config.h
make[5]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
/bin/sh: tools/mkversion: Permission denied
```

julianoes, 2018-12-07:
`git bisect` points somewhere to the point organization:
```
There are only 'skip'ped commits left to test.
The first bad commit could be any of:
abb3817d3121b89e35b6774a4306c86a4616e5cf
cadda8c2630d12fb44cfdc567b68ad251b4f4435
f692ad04d0ea22b8874926e9dc0cac23263bbeae
2d052c02fae3c581a5947f01ace894c54dc1ffbb
We cannot bisect more!
```

dagar, 2018-12-07:
This seems like a race condition in the nuttx part of the build. Almost like it's taking multiple dependency paths to the nuttx setup, but not recognizing that step as a single node and executing it twice.

julianoes, 2018-12-19:
Has it always be the case that the Makefile build is different from the ninja build?
```
ls -l | grep bin                                    
-rwxr-xr-x 1 julianoes julianoes 1021685 Dec 19 09:43 px4_fmu-v2.bin.make
-rwxr-xr-x 1 julianoes julianoes 1021613 Dec 19 09:45 px4_fmu-v2.bin.ninja
```

dagar, 2019-02-03:
Is this still relevant?

julianoes, 2019-02-04:
Yes, the Makefile build is broken as far as I know, unless it has been fixed.

julianoes, 2019-04-08:
I found that the build works with `-j1`.

julianoes, 2019-04-08:
@dagar I'm currently looking at:
```
/home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/Make.defs:36: /home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/.config: No such file or directory
```

and I don't understand why we copy the defconfig file twice:
1. https://github.com/PX4/Firmware/blob/d9f8e37288d8e34baac79b32d71862045946534b/platforms/nuttx/cmake/init.cmake#L111-L112
2. https://github.com/PX4/Firmware/blob/d9f8e37288d8e34baac79b32d71862045946534b/platforms/nuttx/NuttX/CMakeLists.txt#L151

dagar, 2019-04-08:
@julianoes the defconfig now only contains non-default values and needs to be reconstituted at cmake configure time (cmake uses the variables).

 - execute_process does this during configure
 - custom_command does this at runtime if the dependencies change

julianoes, 2019-04-09:
I also found this:

```
Deleting primary custom command output "/home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/.config" because another output "/home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx_config.stamp" does not exist.
Dependee "/home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/CMakeFiles/nuttx_apps_build.dir/DependInfo.cmake" is newer than depender "/home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/CMakeFiles/nuttx_apps_build.dir/depend.internal".
Dependee "/home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/CMakeFiles/CMakeDirectoryInformation.cmake" is newer than depender "/home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/CMakeFiles/nuttx_apps_build.dir/depend.internal".
```

julianoes, 2019-04-09:
And this:
```
Deleting primary custom command output "/home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx/.config" because another output "/home/julianoes/src/Firmware/build/px4_fmu-v4_default/NuttX/nuttx_config.stamp" does not exist.
```

julianoes, 2019-04-09:
Alright, I think this first deletion happens presumably because the file exists from the configure step. It only happens once and at the beginning so I don't think we have to worry about it.

Now, what I found is that the config copy steps are done multiple times if we have `-j4` and not `-j1`, so as you @dagar wrote above there are multiple dependency paths.
I have a really hard to time to understand all the dependencies in this file: https://github.com/PX4/Firmware/blob/d9f8e37288d8e34baac79b32d71862045946534b/platforms/nuttx/NuttX/CMakeLists.txt
Is there any way we could improve it and make it easier to understand as well as prevent make or cmake from getting confused?

julianoes, 2019-04-09:
I believe we're running into this: https://gitlab.kitware.com/cmake/cmake/issues/17585

julianoes, 2019-04-09:
I have it compiling again, I'll make a pull request soon.

PR #11821 (2019-04-11): cmake: fix races without ninja
This fixes build races which happened if "Unix Makefiles" instead of ninja-build was used as the cmake backend.

For any dependencies of commands on files we need to create a target.
Otherwise, if "Unix Makefiles" are used as the generator the commands are run in parallel on the different files which often can lead to races or redundancies in our build.

A nice write-up can be found here:
https://samthursfield.wordpress.com/2015/11/21/cmake-dependencies-between-targets-and-files-and-custom-commands/#custom-commands-and-parallel-make

Fixes #11003.
closed 2019-04-11 by julianoes