PX4/PX4-Autopilot #11004: make px4_sitl gazebo_standard_vtol Error

lyhbuaa, 2018-12-07:
I am using ubuntu 16.04 and gazebo 7. I have finished the jmavsim simulation， now i want to try the vtol with gazebo， i used the command `as follows and failed

> make px4_sitl gazebo_standard_vtol

> 
> FAILED: cd /home/lyh/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build . && /usr/bin/cmake -E touch /home/lyh/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build
> ninja: error: '/home/lyh/src/Firmware/Tools/sitl_gazebo/PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND', needed by 'Imu.pb.cc', missing and no known rule to make it
> ninja: build stopped: subcommand failed.
> Makefile:182: recipe for target 'px4_sitl' failed
> make: *** [px4_sitl] Error 1

TSC21, 2018-12-08:
You do not have protobuf installed. Issue:
```sh
sudo apt-get install libprotobuf-dev libprotoc-dev protobuf-compiler
```

closed 2018-12-08 by TSC21

shu-bamma, 2020-10-14:
Have a similar issue but the soln provided by @TSC21 did not work. Please help, i have also finished the jmavsim simulation but this is what i get with the gazebo one.
```
shubham@haha:~/src/Firmware$ make px4_sitl gazebo_plane[1/4] Performing build step for 'sitl_gazebo'
ninja: error: '/home/shubham/src/Firmware/Tools/sitl_gazebo/PROTOBUF_PROTOC_EXECUTABLE-NOTFOUND', needed by 'Airspeed.pb.cc', missing and no known rule to make it
FAILED: cd /home/shubham/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build /home/shubham/src/Firmware/build/px4_sitl_default/build_gazebo
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
```
any help will be highly appreciated.