PX4/PX4-Autopilot #11006: Building as ROS Catkin Package Install Broken

jgoppert, 2018-12-07:
**Describe the bug**
In master the catkin package builds but the targets are not correctly installed due to odd hacks on installing all of the targets in the binary directory using the cmake directoy install instead of the cmake target install which catkin expects following standard cmake practices.

**To Reproduce**
```bash
mkdir -p ~/px4_catkin/src
cd ~/px4_catkin_src
git clone git@github.com:PX4/Firmware px4
cd  ~/px4_catkin
catkin build
. ./devel/setup.bash
rosrun px4 px4
[rosrun] Couldn't find executable named px4 below /home/jgoppert/git/catkin_px4/src/px4
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /home/jgoppert/git/catkin_px4/src/px4
[rosrun]   /home/jgoppert/git/catkin_px4/src/px4/boards/px4
[rosrun]   /home/jgoppert/git/catkin_px4/src/px4/Tools/jMAVSim/jMAVlib/src/me/drton/jmavlib/log/px4
[rosrun]   /home/jgoppert/git/catkin_px4/src/px4/src/lib/DriverFramework/framework/px4
[rosrun]   /home/jgoppert/git/catkin_px4/src/px4/platforms/posix/lib/px4
[rosrun]   /home/jgoppert/git/catkin_px4/src/px4/msg/lib/px4
[rosrun]   /home/jgoppert/git/catkin_px4/src/px4/msg/templates/px4
```

**Expected behavior**
```bash
rosrun px4 px4

```
This should work but the executable is not found.

dagar, 2018-12-08:
I believe it's all here if you'd like to fix it - https://github.com/PX4/Firmware/blob/master/platforms/posix/CMakeLists.txt#L75-L180

The only thing that needs to be preserved is the creation of a SITL package for running the mission tests. http://ci.px4.io:8080/blue/organizations/jenkins/PX4_misc%2FFirmware-SITL_tests/detail/master/434/pipeline

Any behaviour you'd like to preserve needs at least a very simple test added to the CI system. https://github.com/PX4/Firmware/blob/master/Jenkinsfile#L9-L64

dagar, 2018-12-08:
@lamping7 FYI

TSC21, 2018-12-08:
> Yeah. @TSC21 added it a while back. Getting it back shouldn't be a problem

The only thing I added was a `catkin` (and a `colcon`) build lines, but there isn't any test that's actually running inside a ROS workspace. And that's what me and @jgoppert have been trying to fix past couple of days. PX4 Firmware builds inside a `catkin` workspace but it's not usable whatsoever in a ROS environment

TSC21, 2018-12-08:
> I believe it's all here if you'd like to fix it - https://github.com/PX4/Firmware/blob/master/platforms/posix/CMakeLists.txt#L75-L180

@dagar We have been working around those same lines but seems that every change we do that has repercussions and creates a cascade of errors throughout the generated code we are trying to build. It seems that setting the `PX4_BINARY_DIR` as we currently do has a negative effect for `catkin` in what concerns to absolute vs relative paths.

Your input is welcomed here. Thanks!

jgoppert, 2018-12-12:
I am at a point I need to get this working so I'll spend some time looking into it.

jgoppert, 2018-12-12:
@dagar It is the CMAKE_RUNTIME_OUTPUT_DIRECTORY. Do we need to do this?

PR #11022 (2018-12-13): Fix catkin/ROS build paths
Closes #11006

Need to check before this is merged:

## TODO

* [x] Make sure doesn't break packaging/paths for make commands in non-ROS workflow.
* [ ] Add CI
closed 2018-12-13 by TSC21