PX4/PX4-Autopilot #11008: make px4_sitl_default gazebo error

NgJeffrey, 2018-12-08:
I'm using protobuf version 3.6.0 and when I ran `make px4_sitl_default gazebo` I get this error.
Has anyone dealt with this issue before?

Here is a snippet of the error message:
/usr/include/gazebo-9/gazebo/msgs/subscribe.pb.h:275:87: error: no matching function for call to ‘google::protobuf::internal::ArenaStringPtr::GetNoArena(const string*) const’
   return host_.GetNoArena(&::google::protobuf::internal::GetEmptyStringAlreadyInited());
                                                                                       ^
In file included from /usr/include/ignition/msgs1/ignition/msgs/altimeter.pb.h:23:0,
                 from /usr/include/ignition/msgs1/ignition/msgs/MessageTypes.hh:31,
                 from /usr/include/ignition/msgs1/ignition/msgs/Utility.hh:29,
                 from /usr/include/ignition/msgs1/ignition/msgs.hh:25,
                 from /usr/include/ignition/transport4/ignition/transport/Node.hh:32,
                 from /usr/include/gazebo-9/gazebo/sensors/CameraSensor.hh:22,
                 from /usr/include/gazebo-9/gazebo/sensors/DepthCameraSensor.hh:23,
                 from /home/hades/Firmware/Tools/sitl_gazebo/src/gazebo_irlock_plugin.cpp:23:
/usr/local/include/google/protobuf/arenastring.h:285:31: note: candidate: const string& google::protobuf::internal::ArenaStringPtr::GetNoArena() const
   inline const ::std::string& GetNoArena() const { return *ptr_; }
                               ^~~~~~~~~~
/usr/local/include/google/protobuf/arenastring.h:285:31: note:   candidate expects 0 arguments, 1 provided
In file included from /usr/include/gazebo-9/gazebo/msgs/MessageTypes.hh:113:0,
                 from /usr/include/gazebo-9/gazebo/msgs/msgs.hh:35,
                 from /usr/include/gazebo-9/gazebo/sensors/Sensor.hh:33,
                 from /usr/include/gazebo-9/gazebo/sensors/CameraSensor.hh:24,
                 from /usr/include/gazebo-9/gazebo/sensors/DepthCameraSensor.hh:23,
                 from /home/hades/Firmware/Tools/sitl_gazebo/src/gazebo_irlock_plugin.cpp:23:

Here is the full error message:
https://gist.github.com/NgJeffrey/ccd98762022d72a9cfb193d372094f87

dagar, 2018-12-12:
@TSC21 any idea what's happening here? I'm wondering if sitl_gazebo cmake should be more opinionated about required package versions.

TSC21, 2018-12-12:
This does not seem to be related with `sitl_gazebo`. This seems to be a protobuf problem and I did a quick search online and found similar errors. @NgJeffrey how did you install Gazebo 9 and protobuf?

TSC21, 2018-12-12:
`protobuf` should be installed using `apt-get install libprotobuf-dev`.

NgJeffrey, 2018-12-13:
@TSC21 Gazebo 9 came with ROS if I'm not mistaken. As for protobuf, I downloaded the zip file and did all the usual make commands... I did this several times with different versions and am not sure why its not working.

Moving on, I uninstalled and deleted all traces of protobuf/protoc. Taking your advice, I did `sudo apt-get install libprotobuf-dev`, and it shows that it is already installed
libprotoc-dev/bionic,now 3.0.0-9.1ubuntu1 amd64 [installed,automatic]

However, when I do the sitl command, I get an error stating that I don't have protobuf installed. Is there anything I have to do for the makefile to see that protobuf is installed?

<br>
Here is the full error output
<br>

make px4_sitl_default gazebo
ninja: Entering directory `/home/hades/Firmware/build/px4_sitl_default'
[3/7] Performing configure step for 'sitl_gazebo'
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/hades/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/hades/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/hades/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
-- install-prefix: /usr/local
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
**CMake Error at /usr/share/cmake-3.10/Modules/FindProtobuf.cmake:425 (file):
  file STRINGS file "/usr/local/include/google/protobuf/stubs/common.h"
  cannot be read.**
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:165 (find_package)
  CMakeLists.txt:31 (find_package)

CMake Error at /usr/share/cmake-3.10/Modules/FindProtobuf.cmake:431 (math):
  math cannot parse the expression: " / 1000000": syntax error, unexpected
  exp_DIVIDE, expecting exp_PLUS or exp_MINUS or exp_OPENPARENT or exp_NUMBER
  (2)
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:165 (find_package)
  CMakeLists.txt:31 (find_package)

CMake Error at /usr/share/cmake-3.10/Modules/FindProtobuf.cmake:432 (math):
  math cannot parse the expression: " / 1000 % 1000": syntax error,
  unexpected exp_DIVIDE, expecting exp_PLUS or exp_MINUS or exp_OPENPARENT or
  exp_NUMBER (2)
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:165 (find_package)
  CMakeLists.txt:31 (find_package)

CMake Error at /usr/share/cmake-3.10/Modules/FindProtobuf.cmake:433 (math):
  math cannot parse the expression: " % 1000": syntax error, unexpected
  exp_MOD, expecting exp_PLUS or exp_MINUS or exp_OPENPARENT or exp_NUMBER
  (2)
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:165 (find_package)
  CMakeLists.txt:31 (find_package)

-- Boost version: 1.65.1
-- Looking for OGRE...
[... 53 lines cut ...]
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:191 (find_package)
  CMakeLists.txt:31 (find_package)

-- Could NOT find Protobuf: Found unsuitable version "..", but required is at least "2.3.0" (found /usr/local/lib/libprotobuf.so;-lpthread;-lpthread)
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:191 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/ignition-transport4/ignition-transport4-config.cmake

  but it set ignition-transport4_FOUND to FALSE so package
  "ignition-transport4" is considered to be NOT FOUND.  Reason given by
  package:

  ignition-transport4 could not be found because dependency IgnProtobuf could
  not be found.

Call Stack (most recent call first):
  CMakeLists.txt:31 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/hades/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/hades/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
ninja: build stopped: subcommand failed.
Makefile:171: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

NgJeffrey, 2018-12-25:
I still could not figure out the issue. I suspect that it was a compatibility issue since I have ROS Melodic instead of Kinematic. I made another partition and ran the setup script. That seemed to solve the issue. Thanks for helping me out in the Slack channel.

closed 2018-12-25 by NgJeffrey

PRAYAGSHARMA, 2019-10-16:
> I still could not figure out the issue. I suspect that it was a compatibility issue since I have ROS Melodic instead of Kinematic. I made another partition and ran the setup script. That seemed to solve the issue. Thanks for helping me out in the Slack channel.

hi!
can you specify "made another partition".....means exactly ?

NgJeffrey, 2019-10-22:
Hello, 

If you are having this issue, I would recommend creating another copy of Linux that is compatible with Px4. 

I believe Ros Kinetic supports Px4: http://wiki.ros.org/kinetic
I have Ubuntu 16.04 with Ros Kinetic.

I hope that helps.