PX4/PX4-Autopilot #11009: PX4Flow sonar chosen over Lidar Lite v3 in FMUv3

glegarda, 2018-12-09:
Hello,

I believe there might be an issue with the latest version of the FMUv3 firmware, which is causing my system to use sonar data over lidar data.

When I install the latest stable release with QGC, the console shows the expected output from my Lidar Lite v3 (with PWM interface). However, when I manually upload the latest version of FMUv3, it is the data from the PX4Flow sonar that I get, not that from the lidar. If I unplug the PX4Flow, I get lidar data again.

I am not that familiar with the code myself, so feel free to ask for any output I might be getting from the console that could help you pinpoint the source of the problem.

LorenzMeier, 2018-12-09:
@dpettenuzzo Did we maybe alter boot priorities as part of the latest flow improvements?

dagar, 2018-12-09:
The ordering changed with the introduction of a board defined rc file (eg https://github.com/PX4/Firmware/blob/master/boards/px4/fmu-v3/init/rc.board) that's executed before the regular rc.sensors. For some reason px4flow start was being done per board and not generally.

I'll move px4flow start to rc.sensors after the range sensors are started.

glegarda, 2018-12-09:
@LorenzMeier @dagar thank you.

ecmnet, 2018-12-16:
@dagar Seems also to be the case for fmu-v5 with Sf1xx. (Current master)

dagar, 2019-01-01:
@glegarda @ecmnet could you give master another try? This should be fixed by https://github.com/PX4/Firmware/pull/11123.

PR #11123 (2019-01-01): px4flow start in rc.sensors instead of per board
 - fixes #11009
commit 1b8c2c82 (2019-01-01): px4flow start in rc.sensors instead of per board (#11123)

- fixes #11009
closed 2019-01-01 by dagar