PX4/PX4-Autopilot #11010: EKF2_Replay cant work

Toothsmile, 2018-12-09:
Hi,
I followed the  [system_wide_replay]((https://dev.px4.io/en/debug/system_wide_replay.html)),step by step used EKF2_replay,
i have set:
```
export replay_mode=ekf2
export replay=<abs_path_to_log.ulg>
make px4_sitl none

```
but cant to EKF2_replay,so i have  2 question.
one:
i have a question by the followed setting，i have found sensor_combined dont have mag data,how can this these data  do EKF2.
```
For replay, we only want the replay module to be able to publish the previously identified list of topics. So for replaying ekf2, the rules file looks like this:
restrict_topics: sensor_combined, vehicle_gps_position, vehicle_land_detected
module: replay
ignore_others: true

```
second:
i do EKF2_Replay but the _replay.ulg dont have any data,the folloew is my run result.
```
INFO  [logger] logger started (mode=all)
WARN  [logger] logger: failed to add topic. Too many subscriptions
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2018-12-09/15_41_58_replayed.ulg
INFO  [replay] using replay log file: /home/toothsmile/Study/InterNav/Firmware/Tools/ecl_ekf/log_3.ulg
INFO  [replay] Ekf2 replay mode
INFO  [replay] Applying params from ULog file...
INFO  [px4] Startup script returned successfully
pxh> INFO  [replay] Applying override params from ./replay_params.txt...
INFO  [replay] Replay in progress...
WARN  [replay] Formats for battery_status don't match. Will ignore it.
WARN  [replay]  Internal format: uint64_t timestamp;float voltage_v;float voltage_filtered_v;float current_a;float current_filtered_a;float average_current_a;float discharged_mah;float remaining;float scale;float temperature;int32_t cell_count;float[4] voltage_cell_v;float max_cell_voltage_delta;uint16_t capacity;uint16_t cycle_count;uint16_t run_time_to_empty;uint16_t average_time_to_empty;uint16_t serial_number;bool connected;bool system_source;uint8_t priority;bool is_powering_off;uint8_t warning;uint8_t[5] _padding0;
WARN  [replay]  File format    : uint64_t timestamp;float voltage_v;float voltage_filtered_v;float current_a;float current_filtered_a;float average_current_a;float discharged_mah;float remaining;float scale;float temperature;int32_t cell_count;uint16_t capacity;uint16_t cycle_count;uint16_t run_time_to_empty;uint16_t average_time_to_empty;uint16_t serial_number;bool connected;bool system_source;uint8_t priority;bool is_powering_off;uint8_t warning;uint8_t[1] _padding0;
WARN  [replay] Formats for estimator_status don't match. Will ignore it.
WARN  [replay]  Internal format: uint64_t timestamp;float[24] states;float[3] vibe;float[24] covariances;uint32_t control_mode_flags;float pos_horiz_accuracy;float pos_vert_accuracy;float mag_test_ratio;float vel_test_ratio;float pos_test_ratio;float hgt_test_ratio;float tas_test_ratio;float hagl_test_ratio;float beta_test_ratio;float time_slip;uint16_t gps_check_fail_flags;uint16_t filter_fault_flags;uint16_t innovation_check_flags;uint16_t solution_status_flags;uint8_t n_states;bool pre_flt_fail;uint8_t health_flags;uint8_t timeout_flags;uint8_t[4] _padding0;
WARN  [replay]  File format    : uint64_t timestamp;float[24] states;float n_states;float[3] vibe;float[24] covariances;uint32_t control_mode_flags;float pos_horiz_accuracy;float pos_vert_accuracy;float mag_test_ratio;float vel_test_ratio;float pos_test_ratio;float hgt_test_ratio;float tas_test_ratio;float hagl_test_ratio;float beta_test_ratio;float time_slip;uint16_t gps_check_fail_flags;uint16_t filter_fault_flags;uint16_t innovation_check_flags;uint16_t solution_status_flags;uint8_t nan_flags;uint8_t health_flags;uint8_t timeout_flags;bool pre_flt_fail;
WARN  [replay] Formats for system_power don't match. Will ignore it.
WARN  [replay]  Internal format: uint64_t timestamp;float voltage5v_v;float voltage3v3_v;uint8_t v3v3_valid;uint8_t usb_connected;uint8_t brick_valid;uint8_t usb_valid;uint8_t servo_valid;uint8_t periph_5v_oc;uint8_t hipower_5v_oc;uint8_t[1] _padding0;
WARN  [replay]  File format    : uint64_t timestamp;float voltage5V_v;float voltage3V3_v;uint8_t v3v3_valid;uint8_t usb_connected;uint8_t brick_valid;uint8_t usb_valid;uint8_t servo_valid;uint8_t periph_5V_OC;uint8_t hipower_5V_OC;uint8_t[1] _padding0;
WARN  [replay] Formats for vehicle_gps_position don't match. Will ignore it.
WARN  [replay]  Internal format: uint64_t timestamp;uint64_t time_utc_usec;int32_t lat;int32_t lon;int32_t alt;int32_t alt_ellipsoid;float s_variance_m_s;float c_variance_rad;float eph;float epv;float hdop;float vdop;int32_t noise_per_ms;int32_t jamming_indicator;float vel_m_s;float vel_n_m_s;float vel_e_m_s;float vel_d_m_s;float cog_rad;int32_t timestamp_time_relative;float heading;float heading_offset;uint8_t fix_type;bool vel_ned_valid;uint8_t satellites_used;uint8_t[5] _padding0;
WARN  [replay]  File format    : uint64_t timestamp;uint64_t time_utc_usec;int32_t lat;int32_t lon;int32_t alt;int32_t alt_ellipsoid;float s_variance_m_s;float c_variance_rad;float eph;float epv;float hdop;float vdop;int32_t noise_per_ms;int32_t jamming_indicator;float vel_m_s;float vel_n_m_s;float vel_e_m_s;float vel_d_m_s;float cog_rad;int32_t timestamp_time_relative;uint8_t fix_type;bool vel_ned_valid;uint8_t satellites_used;uint8_t[5] _padding0;
WARN  [replay] Formats for vehicle_status don't match. Will ignore it.
WARN  [replay]  Internal format: uint64_t timestamp;uint32_t onboard_control_sensors_present;uint32_t onboard_control_sensors_enabled;uint32_t onboard_control_sensors_health;uint8_t nav_state;uint8_t arming_state;uint8_t hil_state;bool failsafe;uint8_t system_type;uint8_t system_id;uint8_t component_id;bool is_rotary_wing;bool is_vtol;bool vtol_fw_permanent_stab;bool in_transition_mode;bool in_transition_to_fw;bool rc_signal_lost;uint8_t rc_input_mode;bool data_link_lost;bool high_latency_data_link_active;uint8_t data_link_lost_counter;bool engine_failure;bool mission_failure;uint8_t failure_detector_status;
WARN  [replay]  File format    : uint64_t timestamp;uint32_t onboard_control_sensors_present;uint32_t onboard_control_sensors_enabled;uint32_t onboard_control_sensors_health;uint8_t nav_state;uint8_t arming_state;uint8_t hil_state;bool failsafe;uint8_t system_type;uint8_t system_id;uint8_t component_id;bool is_rotary_wing;bool is_vtol;bool vtol_fw_permanent_stab;bool in_transition_mode;bool in_transition_to_fw;bool rc_signal_lost;uint8_t rc_input_mode;bool data_link_lost;bool high_latency_data_link_active;uint8_t data_link_lost_counter;bool engine_failure;bool mission_failure;uint8_t[1] _padding0;
WARN  [replay] Formats for vehicle_status_flags don't match. Will ignore it.
WARN  [replay]  Internal format: uint64_t timestamp;bool condition_calibration_enabled;bool condition_system_sensors_initialized;bool condition_system_hotplug_timeout;bool condition_system_returned_to_home;bool condition_auto_mission_available;bool condition_global_position_valid;bool condition_home_position_valid;bool condition_local_position_valid;bool condition_local_velocity_valid;bool condition_local_altitude_valid;bool condition_power_input_valid;bool condition_battery_healthy;bool circuit_breaker_engaged_power_check;bool circuit_breaker_engaged_airspd_check;bool circuit_breaker_engaged_enginefailure_check;bool circuit_breaker_engaged_gpsfailure_check;bool circuit_breaker_flight_termination_disabled;bool circuit_breaker_engaged_usb_check;bool circuit_breaker_engaged_posfailure_check;bool offboard_control_signal_found_once;bool offboard_control_signal_lost;bool offboard_control_set_by_command;bool offboard_control_loss_timeout;bool rc_signal_found_once;bool rc_input_blocked;bool rc_calibration_valid;bool vtol_transition_failure;bool usb_connected;uint8_t[4] _padding0;
WARN  [replay]  File format    : uint64_t timestamp;bool condition_calibration_enabled;bool condition_system_sensors_initialized;bool condition_system_hotplug_timeout;bool condition_system_returned_to_home;bool condition_auto_mission_available;bool condition_global_position_valid;bool condition_home_position_valid;bool condition_local_position_valid;bool condition_local_velocity_valid;bool condition_local_altitude_valid;bool condition_power_input_valid;bool circuit_breaker_engaged_power_check;bool circuit_breaker_engaged_airspd_check;bool circuit_breaker_engaged_enginefailure_check;bool circuit_breaker_engaged_gpsfailure_check;bool circuit_breaker_flight_termination_disabled;bool circuit_breaker_engaged_usb_check;bool circuit_breaker_engaged_posfailure_check;bool offboard_control_signal_found_once;bool o [... cut ...]
WARN  [replay]  File format    : uint64_t timestamp;float x;float y;float z;float yaw;float yawspeed;float vx;float vy;float vz;float acc_x;float acc_y;float acc_z;float[3] thrust;
INFO  [replay] Dropout in replayed log, 67 ms
INFO  [replay] Dropout in replayed log, 0 ms
INFO  [replay] Dropout in replayed log, 161 ms
INFO  [replay] Replay done (published 261 msgs, 1.332 s)
INFO  [replay] 
INFO  [replay] Topic, Num Published, Num Error (no timestamp match found):

```
so can tell what i have set wrong or the replay have some wrong?
Thanks!

bkueng, 2018-12-10:
You are trying to replay a log built with an older Firmware with different uORB message definitions. You can only replay that file if you checkout the same Firmware version as the log was recorded with (or a version where the relevant uORB topics have the same definition).

Toothsmile, 2018-12-11:
Thanks for @bkueng replay, i  successful replay EKF2 by the followd set:
`git checkout v1.8.2;make upload my machine`
adjust parameter will cause different result, it can start my EKF Study, thanks anagin, and i have 2 suggest about ekf2 document:
first,i have found the replay module used vehicle_magnetometer for mag estimation，but the [system_wide](https://dev.px4.io/en/debug/system_wide_replay.html) about ekf_replay tell us sensor_combined which have mag,acl, gyro data ,but just have mag and acl data , if can adjuest will make us more clearer.
second, its above sensor_combined too,  [EKF_replay matlab code](https://github.com/PX4/ecl/tree/master/EKF/matlab/EKF_replay), the matlab code's readme file tell us the problem same as the the first question,if it adjuset will be more fine.
thanks !

closed 2018-12-11 by Toothsmile