PX4/PX4-Autopilot #11016: can't find serial configuration parameter MAV_X_CONFIG

NGU12138, 2018-12-11:
Hi all:
      when i use pixhackv3 with firmware v3, i want to config serial port, but i can't find parameter MAV_X_CONFIG in document. Look forward to your reply.
[image]

hamishwillee, 2018-12-11:
That parameter is in master. If you are using a release version that won't be present. What version are you using? PROBABLY you need to set the value of SYS_COMPANION.

@dagar We need to get versioned releases of docs out!

NGU12138, 2018-12-11:
> master
thank you very much. i'm a novice, how can i get master version, in document it is not recomended to change SYS_COMPANION,but generic serial configuration parameters. i want to communicate with companion computer use telem2 or serial4. wish your reply!

NGU12138, 2018-12-11:
> That parameter is in master. If you are using a release version that won't be present. What version are you using? PROBABLY you need to set the value of SYS_COMPANION.
> 
> @dagar We need to get versioned releases of docs out!

closed 2018-12-11 by NGU12138

hamishwillee, 2018-12-11:
@NGU12138 "Master" is the main source code branch that is being developed. It is not stable and may have bugs. YOu basically have to set up a developer environment and build this yourself - see http://dev.px4.io/en/setup/dev_env.html

If you just want to set up an existing drone configuration then perhaps read this old version of the Pixhawk companion doc: https://github.com/PX4/Devguide/blob/8aefbaad8e065ce096d1ef50f07186fdfbff1d06/en/companion_computer/pixhawk_companion.md

We hope to have "versions" of the docs soon, so that you will be able to find the right version of instructions for the Firmware you are using.

NGU12138, 2018-12-11:
> @NGU12138 "Master" is the main source code branch that is being developed. It is not stable and may have bugs. YOu basically have to set up a developer environment and build this yourself - see [...]
thank you very much!