PX4/PX4-Autopilot #11017: mavlink parameter issues to review

dagar, 2018-12-11:
- param index changing during sync
    - it appears that some parameters are only marked active lazily and possibly during the first param sync
 - behaviour with network broadcast resulting in parallel parameter sync

dagar, 2018-12-11:
@DonLakeFlyer @dogmaphobic @hamishwillee looking over the mavlink spec I would guess the parameter index is expected to be maintained.
https://mavlink.io/en/messages/common.html#PARAM_VALUE

[image]

Unfortunately this index changes in PX4 as new parameters are marked active. This doesn't need to happen after boot is complete, but at the moment it's possible.

dagar, 2018-12-11:
The majority of this seems to be the commander preflight checks, which often don't run the first time until after the initial parameter sync.

	INFO  [mavlink] send_one: CAL_ACC_PRIME (32 of 560), 
	INFO  [mavlink] send_one: CAL_BARO_PRIME (36 of 560), 
	INFO  [mavlink] send_one: CAL_GYRO_PRIME (45 of 560), 
	INFO  [mavlink] send_one: CAL_MAG_PRIME (62 of 560), 
	INFO  [mavlink] send_one: COM_ARM_EKF_AB (73 of 560), 
	INFO  [mavlink] send_one: COM_ARM_EKF_GB (74 of 560), 
	INFO  [mavlink] send_one: COM_ARM_EKF_HGT (75 of 560), 
	INFO  [mavlink] send_one: COM_ARM_EKF_POS (76 of 560), 
	INFO  [mavlink] send_one: COM_ARM_EKF_VEL (77 of 560), 
	INFO  [mavlink] send_one: COM_ARM_EKF_YAW (78 of 560), 
	INFO  [mavlink] send_one: COM_ARM_IMU_ACC (79 of 560), 
	INFO  [mavlink] send_one: COM_ARM_IMU_GYR (80 of 560), 
	INFO  [mavlink] send_one: COM_ARM_MAG (81 of 560), 
	INFO  [mavlink] send_one: SYS_HAS_BARO (548 of 560), 
	INFO  [mavlink] send_one: SYS_HAS_MAG (549 of 560), 

As a quick improvement (not the right permanent fix) I'll see if we can have these parameters marked active immediately.

dagar, 2018-12-11:
Is QGC checking or monitoring the param count at any point?

DonLakeFlyer, 2018-12-11:
> * param index changing during sync

Parameter index cannot change during sync. Once sync is complete QGC doesn't care about indices.

> Is QGC checking or monitoring the param count at any point?

QGC knows if it doesn't get all the parameters from the initial sync. Not really by checking the count, but by checking for indices for which it didn't get a param. It will then re-request this missed params by index.

dagar, 2018-12-11:
This really isn't okay then. It's not happening intentionally, but at the moment there's actually no mechanism in place that prevents new params from being marked active (resulting in an index change) during the param sync. In retrospect I bet this explains some of the cases where very slow connections are getting stuck on param sync.

If it's relatively easy I'd consider monitoring the `param_count` in PARAM_VALUE (https://mavlink.io/en/messages/common.html#PARAM_VALUE), and aborting/retrying the entire sync if there's a change. I'll look for a more fundamental fix within PX4 itself.

hamishwillee, 2018-12-11:
The

julianoes, 2020-01-30:
I haven't seen this in a while, closing.

closed 2020-01-30 by julianoes