PX4/PX4-Autopilot #11023: Add new Compass  QMC5883

jamming, 2018-12-12:
The Honeywell HMC5883/HMC5983 compasses are discontinued and hardware provider are having to switch to alternatives. The QMC5883 is such an alternative.  It is pin  compatible with HMC5883/HMC5983. But the device adress is different.

I falshed Pixwark MINI with newly 1.8.2 stable  firmware,  but I found that the firmware  don't support QMC5883.

dagar, 2018-12-12:
What are you using with a QMC5883?

jamming, 2018-12-13:
There is a compass in the GPS module as an external  compass.

dagar, 2018-12-13:
Which one specifically? If you can point at one I can purchases I'll look at adding this driver.

Seeelefant, 2018-12-17:
Try this one [https://www.aliexpress.com/item/GY-273-QMC5883-Module-Triple-Axis-Compass-Magnetometer-Sensor-3V-5V/32893077601.html?spm=2114.search0104.3.1.103d638b2z2mgS&ws_ab_test=searchweb0_0,searchweb201602_1_10320_10065_10068_10547_319_317_10548_10696_453_10084_454_10083_10618_10304_10307_10820_10821_538_537_10302_536_10843_10059_10884_10887_100031_10319_321_322_10103,searchweb201603_51,ppcSwitch_0&algo_expid=123f8f89-3a5c-444a-a04e-5a0437d2e1c4-0&algo_pvid=123f8f89-3a5c-444a-a04e-5a0437d2e1c4](url)

dlwalter, 2018-12-20:
We are using this [HolyBro Micro m8N GPS module](https://www.getfpv.com/holybro-micro-m8n-gps-module.html) which now has the QMC5883 (website is currently out of date)

I'd like to support this however I can.  I see [Ardupilot already has their driver written](https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_QMC5883L.cpp).  Has anyone made a branch and started work?

bys1123, 2018-12-23:
@dagar @dlwalter Thanks a lot for your work. I think lots of manufacturer need this driver.

dlwalter, 2019-01-17:
I have basic functionality implemented in [this WIP pull request](https://github.com/PX4/Firmware/pull/11140).

The sample rate, range, and other configs are all fixed and that functionality can be added later.  But it does work.

dlwalter, 2019-01-17:
If anyone is adding this to their stable branch there are some changes that need to be made like the inclusion of MAGIOSELFTEST case to the ioctl() function.  It looks like some of those ioctl() cases were removed in the latest master but will give errors during preflight checks on stable 1.8.2

dlwalter, 2019-03-15:
The PR was merged into master

closed 2019-03-16 by dagar