PX4/PX4-Autopilot #11024: Conflicting setup scripts

julianoes, 2018-12-12:
I'm confused about the fact that we seem to have two conflicting setup scripts:

1. https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh
2. https://github.com/PX4/Devguide/tree/master/build_scripts

Which ones are to be used/improved?

dagar, 2018-12-12:
If anyone is doing a fresh setup I'd encourage you to try `Tools/setup` and fill in any missing pieces. Once we're happy with everything we can update the dev guide and containers. I'll probably do a fresh ubuntu 18.04 install on my workstation over christmas.

We could add a `setup.bat` for windows as well.

The side benefit of getting all of this in tree is that the toolchain is versioned with firmware.

julianoes, 2018-12-12:
> The side benefit of getting all of this in tree is that the toolchain is versioned with firmware.

Yes that's better.

lamping7, 2019-01-11:
I just setup a fresh Ubuntu 18 machine manually w/ ROS Melodic -- piecing in the required bits when things failed to see what was actually needed. I don't have anything to add necessarily. Is the script for dev purposes for users? Many things in there aren't need just to launch a simulation. The list I put together is pretty short. I'm thinking about this in relation to #11157. Feel free to ping me on slack also.

hamishwillee, 2019-01-11:
@lamping7 Right now users install the dev environment locally by running the scripts in https://github.com/PX4/Devguide/tree/master/build_scripts. The intention is that they be able to run the same setup scripts as is used by docker, so that we only have to keep the environment configured in one place, it is part of the firmware tree, and we know when it breaks. 
The start of that process was https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh

So "both".

That said, I'd prefer that end-users do not need to install everything needed by docker *if possible*.

lamping7, 2019-01-11:
Well... You can't really have "both" *and* have users only need what is required. I agree with the original issue here and think we need a clearer approach. Maybe a dev setup and a user setup? Why not make these distinct to correspond to the appropriate guides?

hamishwillee, 2019-01-11:
> Well... You can't really have "both" and have users only need what is required. 

That is a preference, not a hard requirement. But in any case I disagree - as there is no reason that the dependences for end users and dev system cannot be split into multple scripts and requirement.txt files.

The idea of having just one dev setup is that this is what gets tested/hammered in every single code change and test sequence. Happy to have a separate user setup, but how would you ensure a similar level of testing? Or are you suggesting this would be a separate script run for every test as well.

hamishwillee, 2019-01-11:
> is there a setup linked in the userguide?

No, only in the devguide:  http://dev.px4.io/en/setup/dev_env_linux.html

The scripts linked from there are the ones in the folder linked at the top of this issue. The scripts are also linked from a manual description of setup one level down: http://dev.px4.io/en/setup/dev_env_linux_ubuntu.html

lamping7, 2019-01-11:
I foresee a hierarchy. Similar to the `common` script. Dev inherits from user and adds additional dependencies.

Nooooo.... user --> dev --> docker. Jenkins needs them all plus some. That way they all get tested.

hamishwillee, 2019-01-11:
Whatever works!

hamishwillee, 2019-08-05:
@dagar Have you tried /ubuntu.sh recently. I just tried it on master with Ubuntu 18.04 LTS (using `. ubuntu.sh --no-nuttx` and am getting missing dependency errors.
```
sudo apt-get install python-jinja2 	
  sudo apt-get install python-empy
  sudo pip install catkin_pkg
... eventually I give up.
```
Am I using it wrong?

dagar, 2019-08-05:
The scripts are installing the dependencies with python 3 now (python 2 is EOL this year). 

https://github.com/PX4/Firmware/blob/cfdabb26d779f2780d317c3327c51d21a7ae7b18/Tools/setup/ubuntu.sh#L88

We might need to be a little forceful to make the PX4 build system (cmake) use python 3 directly.

hamishwillee, 2019-08-05:
I think we might. It is also doing something odd - I keep getting errors `Error: PX4 version too low, expected at least v1.9.0 Check the git tag (current tag: 'v1.6.0rc1')` (even though using latest version.

HOw can we be more forceful? The devguide scripts also misbehaving now, which means that I don't trust any of our installation scripts now. Seems like a good time to get ubuntu.sh working (again) and them move to that.

junwoo091400, 2022-07-27:
Now I don't believe we have any more conflicting scripts: https://github.com/PX4/PX4-user_guide/tree/main/scripts

Correct me if I am wrong though!

closed 2022-07-27 by junwoo091400

tiralonghipol, 2023-04-19:
hi
I am in the process of porting a new board and I still get the error
`Error: PX4 version too low, expected at least v1.9.0
`
when running 'make my_new_board'

julianoes, 2023-04-21:
@tiralonghipol not sure how your question is related. I would suggest to create a new issue and properly describe what you did, what the error is, etc.