PX4/PX4-Autopilot #11028: `wqueue_test` currently broken`

julianoes, 2018-12-12:
When I look at the output of `make tests` I see:
```
wqueue_test: command not found
```

So presumably this test is not actually running right now. Therefore, two things need to be fixed:
1. wqueue_tests needs to be compiled so it can be run.
2. A test should fail if a command is not found (`set -e`?).

dagar, 2018-12-12:
If I was going to spend time here I think adding a bit more structure would help. Something that registers the tests and executes all of them. Then we could push each test into its corresponding module/driver, and slightly untangle `systemcmds/tests`.

Instead of `px4_add_module` we could have an alternate `px4_add_test` that registers the test and generates a simple script that runs all of them sequentially, catches failures, and reports overall success. That same mechanism should work in sitl and on actual hardware allowing us to drop `ROMFS/px4fmu_test`.

julianoes, 2019-03-13:
Still relevant.

julianoes, 2024-03-04:
I feel like this is no longer relevant.

closed 2024-03-04 by julianoes