PX4/PX4-Autopilot #11031: PX4 Flight Review logs has no local_position_X and local_position_Y items.

E666GT, 2018-12-13:
PX4 Flight Review logs has no local_position_X and local_position_Y items.

This is the log.
https://logs.px4.io/plot_app?log=51892e06-b614-4613-9ddf-6dff596c5de2
or
https://logs.px4.io/plot_app?log=3468cb3f-4c61-4592-9f38-013384bc3a20

Every log works fine, but local position X and local position Y remains 0, I don't know why.

Thanks for your help!
[FlightLog20181213_1.pdf](https://github.com/PX4/Firmware/files/2676317/FlightLog20181213_1.pdf)

[image]

Here's the system info:
_Airframe: | DJI Flame Wheel F450Quadrotor x (4011)
Hardware: | PX4FMU_V2 (V2)
Software Version: | v1.8.1 (82aa24ad)
OS Version: | NuttX, v7.22.0
Estimator: | EKF2
Logging Duration: | 0:01:42
Vehicle LifeFlight Time: | 11 minutes 26 seconds
Vehicle UUID: | 001F00323235511034363339
Max Altitude Difference: | 2 m
Max Tilt Angle: | 20.9 deg
Max Rotation Speed: | 168.2 deg/s
Average Current: | 16.6 A
Max Current: | 24.4 A_

bkueng, 2018-12-14:
Your GPS does not have a fix, so there is no position estimate. Do you fly indoors?

E666GT, 2018-12-14:
> Your GPS does not have a fix, so there is no position estimate. Do you fly indoors?
@bkueng 
But before i have a try, I have a question, it means without GPS there will be no estimation of X&Y ?  I don't think so.Other sensors like acceleration will contribute to XY estimation,too. Am I right?

bkueng, 2018-12-14:
Yes they contribute, but you don't have a position estimate w/o GPS

E666GT, 2018-12-14:
Thanks!@bkueng 
I did as you say. I did a test flight outdoor, and it gives me detials in log, including Position X and Position Y.etc. 
[](https://logs.px4.io/plot_app?log=6f917f10-7e73-470d-8417-a6c452f83e9a)
Thanks again!
I'm close this issue.

closed 2018-12-14 by E666GT