PX4/PX4-Autopilot #11033: FMUv4 is out of RAM

davids5, 2018-12-13:
**Describe the bug**
when SYS_AUTOSTART == 0 FMUv4 is out of ram in startup
```
[[boot] Fault Log info File No 4 Length 3177 flags:0x01 state:1
[boot] Fault Log is Armed
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
HW arch: PX4_FMU_V4
FW git-hash: cfad556a1609048029e66f540c7e87e87f2d8905
FW version: 1.9.0 0 (17367040)
FW git-branch: master
OS: NuttX
OS version: Release 7.22.0 (118882559)
OS git-hash: d07a1d459ee9b5b56dbf06a160e8faa4c49fcd10
Build datetime: Dec 13 2018 09:05:30
Build uri: localhost
Toolchain: GNU GCC, 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
PX4GUID: 0001000000003535323035375110003e0039
MCU: STM32F42x, rev. 3
[hardfault_log] Fault Log is Armed
INFO  [tune_control] Publishing standard tune 1
INFO  [dataman] Unknown restart, data manager file '/fs/microsd/dataman' size is 350216 bytes
INFO  [load_mon] stack check enabled
INFO  [param] selected parameter default file /fs/mtd_params
WARN  [rgbled] no RGB led on bus #1
WARN  [blinkm] I2C init failed
WARN  [blinkm] init failed
WARN  [rgbled_ncp5623c] no RGB led on bus #1
nsh: rgbled_pwm: command not found
Board init: /etc/init.d/rc.board
WARN  [drivers_board] reset done, 50 ms
WARN  [hmc5883] no device on bus 1 (type: 2)
WARN  [lis3mdl] no device on bus 2
INFO  [ist8310] no device on bus 1
BMP280_I2C on I2C bus 1 at 0x76 (bus: 100 KHz, max: 100 KHz)
WARN  [bmp280] id of your baro is not: 0x58
WARN  [bmp280] bus option number is 2
ERROR [bmp280] driver start failed
ERROR [bmm150] Internal I2C not available
HMC5883_SPI on SPI bus 1 at 5 (11000 KHz)
WARN  [hmc5883] no device on bus 1 (type: 3)
WARN  [lis3mdl] no device on bus 2
LIS3MDL_SPI on SPI bus 1 at 7 (11000 KHz)
INFO  [lis3mdl] Poll rate set to max (80hz)
WARN  [mpu6000] no device on bus #3 (SPI1)
MPU6000 on SPI bus 1 at 6 (1000 KHz)
INFO  [mpu9250] Bus probed: 2
WARN  [mpu9250] no device on bus 2
INFO  [mpu9250] Bus probed: 3
MPU9250 on SPI bus 1 at 4 (1000 KHz)
INFO  [mpu9250] accel cutoff set to 30.00 Hz
INFO  [mpu9250] gyro cutoff set to 80.00 Hz
INFO  [mavlink] mode: Normal, data rate: 20000 B/s on /dev/ttyS0 @ 921600B
WARN  [mavlink] hardware flow control not supported
px4flow [46:100]
INFO  [px4flow] scanning I2C buses for device..
ERROR [param] Parameter SYS_USE_IO not found
MS5611_SPI on SPI bus 2 at 3 (20000 KHz)
WARN  [bst] no devices found
INFO  [mpu9250] Bus probed: 2
WARN  [mpu9250] no device on bus 2
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
Starting Main GPS on /dev/ttyS3
Starting MAVLink on /dev/ttyS1
INFO  [mavlink] mode: Normal, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
No autostart ID found
INFO  [logger] logger started (mode=all)
nsh: unset: unsetenv failed: Out of memory
```
**nsh: unset: unsetenv failed: Out of memory**

```

bkueng, 2018-12-14:
@davids5 the problem is in NuttX: after the last unset, `unsetenv` will use `newsize==0` in https://github.com/PX4/NuttX/blob/px4_firmware_nuttx-7.22%2B/sched/environ/env_unsetenv.c#L97.
The realloc call then returns NULL in https://github.com/PX4/NuttX/blob/px4_firmware_nuttx-7.22%2B/mm/mm_heap/mm_realloc.c#L100, which then leads to a `ENOMEM` error.

bkueng, 2018-12-20:
@davids5 can you look into this please?

davids5, 2018-12-20:
@bkueng Sorry I should have acknowledged your findings - Thank you for debugging it.  

It is on my list for today! I had an massive fragmentation issue with the upgrade to 7.72 that I am working on. I originally thought this was the genesis until I found the cause that I am working with Greg on resolving.

closed 2018-12-22 by LorenzMeier